prusa3d/Prusa-Firmware #1141: Toshiba Flash Air

Lino77, 2018-09-12:
Is it possible to delay the initialization of the toshiba flashair, have a W04, and this works only when I push it in, about 1 mm pull out sd card appears removed, and then push back into it

mcstoufer, 2018-09-12:
Lino, I've had this problem too in the past, and it appeared to be just a placement issue in the slot. I've seen a few others with this in the Forum. Once you get it set right, do you still see it not showing up on reset?

Lino77, 2018-09-13:
Well it does work, but I get no IP displayed

Thijs74, 2018-09-23:
Same issue here:
- Card inserted, mentioned in display
-WiFi is active and found by other computers
-Pulling out 1-2 mm, and push back in, all is working.

However, when powering down the printer, and powering it up again will make you suffer from the same issue. Honestly I doubt if this has to do by the correct placement in the slot as mentioned by MCStoufer (no offence ;-)) ).

IP Address is never displayed at all.

Kind regards,

Thijs

slate0, 2018-11-09:
I have a W04 which has been working for ~1 month.  If you request I can experiment with it to try to help.

mcstoufer, 2018-11-21:
> Same issue here:
>
> * Card inserted, mentioned in display
>   -WiFi is active and found by other computers
>   -Pulling out 1-2 mm, and push back in, all is working.
>
> However, when powering down the [...]

I have noticed after a FW update to the card, that it does take a noticeably longer time to join to the WiFi AP and be visible by my Mac via SMB. I have not touched the placement of the card in the slot since then.

merlin2105, 2019-05-28:
Hi, the Same issue here on my Prusa MK3S Printer 
Have any one a solution for this...

Card inserted, mentioned in display
-WiFi is active and found by other computers
-Pulling out 1-2 mm, and push back in, all is working.

GreenShadowMaker, 2019-07-27:
I really like how the card is working, except having to remove/reinsert for the printer to recognize it after power-cycle.  Usually, the wifi part IS working and I can transfer data to it when the printer doesn't see it.

rhounsell, 2019-08-26:
See https://github.com/prusa3d/Prusa-Firmware/issues/1685

GunSmoker, 2019-10-20:
Same issue here.

https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/mk3-does-not-recognize-flashair-card/#post-105534

closed 2023-09-06 by github-actions[bot]