prusa3d/Prusa-Firmware #1148: Feature Request:

RomarQuitasol, 2018-09-13:
"Layer Height Pause Option"

I would like to suggest, if there is a possibility to implement the option, that while printing, one can go to "Tune" for example, and then select an option to "Pause" or "filament change". I find this to be convenient because like now, I have just forgotten to do add a change filament option.Now I have to wait for a specific height, pause manually. This is tedious, waiting for this to happen :P. Just a thought/idea that you might like, and what us users would like.

Kind Regards, 

Romar Quitasol

ohthetrees, 2019-06-13:
You mean pause at a particular layer height? As you've written it, I don't see a distinction between a pause in the tune menu, and "pausing manually".

RomarQuitasol, 2019-06-13:
> You mean pause at a particular layer height? As you've written it, I don't see a distinction between a pause in the tune menu, and "pausing manually".

Hi, 

I mean, being able to stipulate a filament change on specific Z height/s, using the knob next to the LCD. So let's say ive already sliced a file. I can then find this file on the sd card and indicate that I'd like to add filament changes at specific heights.

leptun, 2019-06-13:
It would be very complicated as the printer also does z-hop, effectively reaching that layer before actually printing it. Also it would require precious flash space that can be used on something more useful.

You could do this if you were printing using octoprint with a plugin. I think it was called colorchange. It allows you to add filament changes while printing, just like you want it.

closed 2019-10-11 by michalxfanta