prusa3d/Prusa-Firmware #1149: 3.4.0RC1/RC2/Final: Statistic Screen - Filament used weird numerbers

RacingHell, 2018-09-13:
Hi,
since RC1 my statistics screen messed up. Filament used was something like 2xxx m on 3.3.1. Now it shows 65845.60 m. I haven't by faaar printed 200+ spools until now. :)

Any chanche to get that fixed?

maximlevitsky, 2018-09-14:
It does work here, but what doesn't work on RC2 is that during the print, the total print time of current print starts to show an negative number in the statistics screen

maximlevitsky, 2018-09-14:
For your issue, which printer do you have and from which firmware version did you update. I would strongly suspect that something messed up your EEPROM on the printer, since the code which prints the values looks ok to me.

RacingHell, 2018-09-14:
I have a MK3 and updated every released firmware since january. As I told, updating 3.3.1 to 3.4.0 RC1 messed up my stats.

maximlevitsky, 2018-09-14:
The eeprom layout must have changed I guess or something messed it up. We need a way to dump the eeprom through some G code.
Maybe the initial bootloader can be used to dump eeprom too, I don't know.
Anyway, don't think I can help on this yet.

mkbel, 2018-09-20:
It looks like something has written 0x1 at 4083 EEPROM address.

mkbel, 2018-09-20:
It looks like something has written 100 (0x64) at 4083 EEPROM address. It has been experimentally tested on printer. Writing 100 at 4083 EEPROM address caused total filament used statistic jump from 5.09m to 65541.09m.

mkbel, 2018-09-20:
Potential candidates:
extrudemultiply as it is commonly 100 and is written to EEPROM during power panic.

Check following code for underflow:
		if (total_filament_used > ((current_position[E_AXIS] - destination[E_AXIS]) * 100)) { //protection against total_filament_used overflow
			total_filament_used = total_filament_used + ((destination[E_AXIS] - current_position[E_AXIS]) * 100);

RacingHell, 2018-09-24:
> 
> 
> It looks like something has written 100 (0x64) at 4083 EEPROM address. It has been experimentally tested on printer. Writing 100 at 4083 EEPROM address caused total filament used statistic jump from 5.09m to 65541.09m. And would do equivalent for any original value less then 655.36m.

Hmm.. in my case the original value was somthing around 2000 meters, now it's showing 65903.69m. As it was 65845.60m at the initial issue report it's incrementing healthy so far.

Not shure how to explain the difference of 368,69m. 65903.69-65535+2047= 2415,69m -> this could match my real statistics.

mkbel, 2018-10-02:
> Hmm.. in my case the original value was somthing around 2000 meters, now it's showing 65903.69m. As it was 65845.60m at the initial issue report it's incrementing healthy so far.
>
> Not shure how to [...]

So I am out of ideas now. The number reported looked like 2^16 for the first sight, so I suspected something like 16 bit unsigned number underflow. But numbers are stored or operated on with different scaling than 1 meter - 1 centimeter in case of EEPROM storage and 0.01 mm in case of RAM 32 bit unsigned integer. RAM value underflow would cause 42949.67296 meters jump.

Maybe this is completely random value with no pattern, caused by power failure or reset during EEPROM write. Statistics values are not protected against this.

DRracer, 2020-09-22:
@RacingHell did a similar issue occur on later firmware upgrades? If not, we'd like to close this issue.

RacingHell, 2020-09-22:
@DRracer I haven‘t tracked the issue for a long time, since I am actually not that active in printing. Maybe it was just a random bug from updating frequently from up the beginning of new releases.

closed 2020-09-22 by RacingHell