prusa3d/Prusa-Firmware #1152: Slow filament loading in stealth mode

alientek, 2018-09-14:
When in stealth mode loading new filament is very slow!

ironjoe13, 2018-09-17:
+1 for this, in silent mode there is no fast extruder movement to load the filament, it is actually slower than when it gets to the extruding speed

PavelSindler, 2018-09-18:
Can you connect to printer via serial line (Octoprint, Pronterface) and send M503 to the printer and send me here the printer response?

darBis, 2018-09-21:
@PavelSindler: During filament loading operation only the phase during Z axis movement is very slow.
Issue manifest only when printer is in stealth mode. In normal mode filament loading works as expected.

Numbers for stealth mode acceleration limits look crazy. :-) I am capable to do full eeprom dump over ISP if it would be helpful.

Output from M503:
```
Send: M503
Recv: echo:Steps per unit:
Recv: echo:  M92 X100.00 Y100.00 Z400.00 E280.00
Recv: echo:Maximum feedrates - normal (mm/s):
Recv: echo:  M203 X200.00 Y200.00 Z12.00 E120.00
Recv: echo:Maximum feedrates - stealth (mm/s):
Recv: echo:  M203 Xnan Ynan Znan Enan
Recv: echo:Maximum acceleration - normal (mm/s2):
Recv: echo:  M201 X1000 Y1000 Z200 E5000
Recv: echo:Maximum acceleration - stealth (mm/s2):
Recv: echo:  M201 X960 Y960 Z4294967295 E4294967295
Recv: echo:Acceleration: S=acceleration, T=retract acceleration
Recv: echo:  M204 S1000.00 T1250.00
Recv: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Recv: echo:  M205 S0.00 T0.00 B0.00 X10.00 Y10.00 Z0.40 E2.50
Recv: echo:Home offset (mm):
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:PID settings:
Recv: echo:   M301 P22.67 I2.49 D51.60
Recv: echo:PID heatbed settings:
Recv: echo:   M304 P126.13 I4.30 D924.76
Recv: echo:Retract: S=Length (mm) F:Speed (mm/m) Z: ZLift (mm)
Recv: echo:   M207 S3.00 F2700.00 Z0.00
Recv: echo:Recover: S=Extra length (mm) F:Speed (mm/m)
Recv: echo:   M208 S0.00 F480.00
Recv: echo:Auto-Retract: S=0 to disable, 1 to interpret extrude-only moves as retracts or recoveries
Recv: echo:   M209 S0
Recv: echo:Filament settings: Disabled
Recv: ok
```

PavelSindler, 2018-09-24:
You can solve this issue by sending M502 command (load default settings) and then M500 (store current settings to EEPROM). If you had some custom setting, you will lose it (usually PID calibration results). Or you can downgrade or use printer in normal mode for now and wait for 3.4.1 fw, which fixes this bug.

darBis, 2018-10-05:
Firmware upgrade to 3.4.1 solved the issue for me.
Thanks

closed 2018-10-05 by PavelSindler

ayberkozgur, 2018-10-16:
I was having weird behavior with the Z axis being totally non-responsive, then becoming responsive suddenly upon fidgeting in the menu or starting a print. The X and Y axes were somehow responsive all the time. Later, I realized that the Z axis was responding but was unbelievably slow, to the point of almost not moving. An `M503` confirmed the above issue and the firmware update fixed it.