prusa3d/Prusa-Firmware #1154: 3.4 Final:   Doesnt Resume after FIlament runout / undetection

christizzz, 2018-09-15:
I was printing... then the Sensor triggered... the print head moved to the front right corner and the printer asked to unload filamen... after unloading and loagin filament... the print head moved to its last position and sits there and it started a blob on the print and didt move.  In the menu o pressed the resume print... the print head homed all axes really SLOW... and then moved back to  the print  to a wrong position and on a lower Z height wich destroyed the print on its path...i had to reset...

nojan1, 2018-09-16:
Exactly the same thing happened to me. Except the filament got stuck on the spool, so didn't technically run out. Anyhow didn't resume correctly and crashed into the print.

vw671, 2018-09-16:
Yup. exactly the same issue after runout detection. reverted back to 3.3.1

theinternetftw, 2018-09-20:
Same issue.

PavelSindler, 2018-09-21:
Fixed in PR https://github.com/prusa3d/Prusa-Firmware/pull/1174. We will prepare 3.4.1 asap. Thank you for reporting.

KA-ARCHER, 2019-05-21:
Only in part, the print was going fine, it stopped printing the head came to the front right, asked to unload, then reload, went back to printing without problem. The print was OK. In the error statistics no comment, 0 errors.

3d-gussner, 2020-07-13:
@christizzz Is this still an issue? This post have been stale for long time. If it has been fixed please close this issue.
[stale]

closed 2020-07-28 by christizzz