prusa3d/Prusa-Firmware #1156: 3.4.0 Temp Calibration Issues

sarf2k4, 2018-09-15:
Flashed to 3.4.0 on my mk3 did some print and issues with bed leveling where left was too far, right side was squished down, then ran the temp calibration.

Last night is when I ran the temp calibration, printer got up until step 6/6 at 110'c, it got stuck there even for about 3 hours. Just read similar issues here https://github.com/prusa3d/Prusa-Firmware/issues/822 which the temp calibration needs to have pinda reached 60'c heated from the bed then able to complete.

**Issues**
I didn't build an enclosure for my printer and its next to the table I always using, open air I guess. checked the temperature status, from printer's support sub-menu, ambient = ~46'c, pinda = ~58'c. I suspects, as what I read from other threads that at 110'c bed, pinda had to be heated till 60'c. 
My pinda can't reach until 60'c, so it is a problem for those who didn't have enclosure.

christizzz, 2018-09-16:
Ibhave exactly the same problem..git my printer 2 weeks ago with 3.3.1... i still did not find a solution.... i am pretty sure it is firmware relates... I switched the rods ledt and right as well as their holder it dod not influence the left side... I changes my bed variance fron 0.5 to 0.02 and still not a single improvement...

stahlfabrik, 2018-09-16:
There are three better ways to calibrate the temperature compensation of the PINDA probe:
1. 
https://github.com/PrusaOwners/prusaowners/wiki/Manual_print_based_PINDA_temperature_calibratio
n
2.https://github.com/crzcrz/prusaowners/blob/master/First_layer_and_PINDA_temperature_calibration_with_calipers.md
3. 
https://github.com/PrusaOwners/prusaowners/wiki/PINDA_Temperature_Calibration_with_Electrical_Continuity

They all work and are better than the automatic calibration

christizzz, 2018-09-16:
So do you think it is because of pinda calibration ?

stahlfabrik, 2018-09-16:
Maybe - but I was referring to @sarf2k4 who has problems with the automatic PINDA calibration.

If I were you @christizzz I would check the Z tops. Are they the same distance from the heatbed?

christizzz, 2018-09-16:
you mean the parts holding the spindle and smooth rod ?

stahlfabrik, 2018-09-16:
Yes, on calibrate Z the X gantry moves into them. If one is loose or the distance to the bed of both is not the same, the X gantry will not be parallel to the bed

stahlfabrik, 2018-09-16:
@christizzz You could ask in the discord community for rapid back and forth - though I am AFK for now: https://discord.gg/5EXfP4

christizzz, 2018-09-16:
Ok will check this when my long print finishes... any good way to check this ?  I printed some spare Z tops i can try with them tomorrow

stahlfabrik, 2018-09-16:
I would advise you to ask in the discord community

sarf2k4, 2018-09-16:
I also forgot to mention about my question regarding the temp calibration option. What happens if I turned that on from the settings, will it heat up the pinda and then start printing my object just like the old mk2s does by heating up the pinda for 2 minutes?

sarf2k4, 2018-09-17:
I'm sorry, I'm still not used to the discord community yet. 

Which method should I be following, the original method, or the one uses calipers?

I was thinking of getting a good 0.2mm layer height first before doing the manual temp calibration

stahlfabrik, 2018-09-17:
It seems it is a matter of personal taste:-) 
The print based approach has been used by many. The calipers based approach is much newer. Print based approach requires a „feel“ and good eyesight. While the measurements are more direct and scientific. 
In the both methods should give you the offsets that are matching your PINDAs temperature curve.

sarf2k4, 2018-09-17:
I also have another question, what about the formula in that page you linked? 

`usteps(T) = (live_adjust(35) - live_adjust(T)) * 400`

If i'm not mistaken the live_adjust(35) is the temperature
While I'm not sure about live_adjust(T) here. Was it the z-live adjust values in micron?

If we are to calculate based on the printer's z-live adjust value, then the result of ustep(T) varies or 0 (if perfect layer but this should be okay)

RacingHell, 2018-09-17:
Did you consider to check your pinda mount hight already? Maybe lowering it a tiny bit could give it the lasting 2° to reach 60°. Whats your  Z-correct for printing PLA. For me a value of -0.825 ist the sweet spot and calibration executes well every time. What you could check as well ist the temperature homogeneity of your heatbed - if you have the chanche to check it with a thermal cam to make shure the measuring spot gets hot enough and is able to heat up the pinda. Otherwise your pinda might be faulty... ...or the bed sensor.

...sorry for my coding-offtopic! :)

sarf2k4, 2018-09-18:
Hi @RacingHell , 

I also have around 700-800 z-offset. I believe that it is the metal body that dissipate heat faster than what the bed able to heat it up

rongith, 2018-10-08:
See my comment https://github.com/prusa3d/Prusa-Firmware/issues/822#issuecomment-427971171 to reach 60°C with the PINDA.

Prusa-Support, 2022-05-05:
This issue will be closed as obsolete and as a duplicate of issue #822.

Michele Moramarco
Prusa Research

closed 2022-05-05 by Prusa-Support