prusa3d/Prusa-Firmware #1159: Impossible to get an even first layer

christizzz, 2018-09-16:
I have been struggling since 2 weeks now and i cant find a solution to my problem... The plastic is not getting squished on the left side of the bed... and is a bit more squished on the right...

So i started troubleshooting:

1) first i though it is due to the bed and carriage and /or rods... that are warped.... ---> got replacement from prusa to try... didnt help... however i got less vibration with the rods...

2) i checkd the variance of my bed it was around 0.4-0.5mm so i decided to use a mod found here on the forum and to use springs in order to level the bed manually first... so i built the mod... and i got a varianc of 0.02 - 0.05 ( depends on Temps )..which is pretty leveled... the mesh level vuzualiyer showed nearly a constant surface... ---> still getting plastic unsquished on the Left

3) i though i might be hardware... I completely switch the postion of the Smooth Rods AS well as their supports from left to right...--> spoiler alert: it is still not squished on the left...

4) i played with bed level correction -50 on the left and +50 on the right...Nothing

5) and i reflashed firmware... did all calibrations again... still same result...

I noticed that the extruder rises a bit ( by observing the Z spindles) each time it goes past the first row of measurment poiint s( going to the left)

Now i am loosing hope... here are some photos: https://imgur.com/a/WCCqdzo

and here is the file iam trying to print: https://www.thingiverse.com/thing:1776031

slate0, 2018-11-09:
I am seeing a very similar problem with my first layer.  That picture looks a lot like my results, and mine get worse at times.

I documented all of my details here: https://shop.prusa3d.com/forum/assembly-and-first-prints-troubleshooting-f62/sudden-first-layer-problems-t25775.html

slate0, 2018-11-13:
In the end, I think my PEI sheet was getting worn.  When filament stopped sticking, I lowered my first layer towards my bed.  Then I got symptoms of "too close" in addition to "too far".  A quick wipe with acetone has me back to satisfactory prints (not perfect).

Prusa-Support, 2022-05-20:
Hello! I am sorry for intervening on this so late. Was the issue resolved in the meantime? 
I have to point out that acetone wouldn't be possible to use on textured or satin sheet (I saw from the picture that you had textured sheet), as it would damage the textured-satin sheet. OK to use on smooth sheet once in while, though. 
In case the issue was still occurring, was it possible to correct it with the Bed Level Correction? https://help.prusa3d.com/article/bed-level-correction_2267

Let me know, in case of anything to review we can review it with me via email info@prusa3d.com mentioning me ("attn: Alessandro"). Closing the issue for now here. 

Alessandro Pantaleo
Prusa Research

closed 2022-05-20 by Prusa-Support