prusa3d/Prusa-Firmware #1161: Using the i3 MK3 with a UPS

stanzlavos, 2018-09-17:
I recently purchased a 1100VA/660W UPS for my i3 MK3. I tested it multiple times by turning off the power. And the MK3 always resets. :( I had a chat with customer support and I was told that the only way is to disable "power panic" (by removing the power panic cable) from the EINSY if I need to use a UPS.

Is there no other solution ? I would like to retain the power panic feature as well.

jettoblack, 2018-09-17:
I use a MK3 with a UPS and it works fine...

> I tested it multiple times by turning off the power. And the MK3 always resets.

Exactly where and how are you turning off the power? And when you say the MK3 resets, do you mean it remains on but the control board reboots? Does it show the welcome screen on the LCD? Or it turns off? Or something else?

> Another problem I see is that, when on the UPS, nozzle/bed heating stops and it resumes as soon as I turn the power back on.

I don't see how, since the MK3 has no way of knowing about the UPS. Power panic shouldn't be kicking in if the UPS is working correctly, the MK3 will never see any power drop or blip.

I suggest you test your UPS using a lamp or something like that to make sure it stays on when switching to battery. If it blinks even briefly, maybe there's something wrong with the UPS.

stanzlavos, 2018-09-17:
> Exactly where and how are you turning off the power? And when you say the MK3 resets, do you mean it remains on but the control board reboots? Does it show the welcome screen on the LCD? Or it turns off? Or something else?

I just turn the switch to the socket feeding the UPS off. And the MK3 resets, comes back to the welcome screen. If it is in the middle of a print, it tries to recover the print.

> I don't see how, since the MK3 has no way of knowing about the UPS. Power panic shouldn't be kicking in if the UPS is working correctly, the MK3 will never see any power drop or blip.

I have absolutely no idea. Maybe the PSU detects a voltage drop (when on the UPS, as it is loaded) ? I had a laptop (a G73JH with a dead battery ;) ) connected to the same UPS. I did perform tests with and without the laptop connected. Unfortunately, I don't remember if the laptop was connected when I saw the nozzle temperature issue. 

And to complicate the problem a little more : the UPS I used for this test was a 1500VA one. :( Here, the reset when turning the power off was not so frequent (so yes, the more expensive UPS has a faster switching time - "6ms typical : 10ms maximum" vs "10ms typical : 12ms maximum").

Anyhow, I am certain of one thing : 

- MK3 running with power
- Pre-heat PLA
- Temp starts to rise
- Turn off power to UPS
- No reset
- Temp starts to fall
- Turn on power to UPS
- Temp starts to rise

We could leave this for the time being though. ;)

> I suggest you test your UPS using a lamp or something like that to make sure it stays on when switching to battery. If it blinks even briefly, maybe there's something wrong with the UPS.

I have a UPS that shows the load. When printing (bed/nozzle is heating), it reads around 360W. I have a gaming rig that goes to around 360-375W while gaming. My desktop stays on with no problems on the same UPS.

And more importantly : After I removed the power panic cable, no more resets and everything works fine (on the 1100VA UPS) just as the customer support guy said. :) So the question is, how can I retain power panic and still use a UPS.

jettoblack, 2018-09-18:
Do either of your UPS models have Active PFC and/or true sine wave features? Mine does. The stepped sine waves from a UPS without this feature may be enough to trigger the power panic board. I'm on 110V 60Hz, if you are on 50Hz that could make a difference too.

stanzlavos, 2018-09-18:
I am from India and we are on 220V/50Hz. I couldn't find out if my UPS (APC BX1100C-IN) has active PFC. Some technical info that might me useful :

Waveform type : Stepped approximation to a sinewave
Input frequency : 45 - 65 Hz (auto sensing)

Anyhow, it does work fine with my computer. So I am assuming that there is nothing wrong with the UPS itself.

jettoblack, 2018-09-18:
> Waveform type : Stepped approximation to a sinewave

That's probably the problem. A stepped (square) wave works ok for powering devices with a AC-DC power supply like a computer, but to the power panic circuit, the square wave can look like a sudden loss of AC. If you can try a true sine wave output UPS, it might help.

AnHardt, 2018-09-18:
There are several types of UPS.
* online UPS: Always loads the battery. The secondary AC side is always fed by the battery. Advantage: very clean output even in the moment power gets lost. Disadvantage: High losses.
* rest of the world including smart UPS: Battery is loaded when needed. Secondary AC side is usually connected to the primary side. When power drops the secondary is switched to the converter. Advantage: Less losses. Disadvantage: Not that clean transition.

stanzlavos, 2018-09-18:
Mine is not an online UPS. It is line interactive. The online ones are expensive. :D

Pathogenius, 2018-09-25:
I have read about this issue several times, and each time it does happen to be the case the UPS is not of suitable caliber. I recall that the solution was to use a UPS that provides PURE SINE WAVE. These UPS can be had for cheap. A UPS with poor electronics might be best used for things like coffee bean grinders and incandescent lights.

Crunch69, 2018-10-10:
I reported this issue many months ago issue #898 back in July https://github.com/prusa3d/Prusa-Firmware/issues/898.  Using a APC 1500watt.  Note it only happens with a Prusa Mk3  UPS works perfectly with a Creality CR-10

dmoralesdev, 2019-05-02:
> I use a MK3 with a UPS and it works fine...
>
>
> Exactly where and how are you turning off the power? And when you say the MK3 resets, do you mean it remains on but the control board reboots? Does it [...]

@jettoblack what UPS brand/model/capacity are you using. I would like to buy a UPS too but I have been unable to confirm the exact power requirements of the MK3S. Thanks.

Kachidoki2807, 2020-02-16:
> Another problem I see is that, when on the UPS, nozzle/bed heating stops and it resumes as soon as I turn the power back on.

I have this behavior too, tested today. I use the same offline UPS, an old Infosec SL500 since I had an MK2S, which worked flawless excepted that I got a printer reboot once while printing ABS. I assume that the capacitor inside the power supply couldn't sustain the power during the switch time.

On my MK3, FW 3.9.0 beta, the printer stay powered up, but the heaters stops.
Given the price of an online UPS, I prefer to lose a print job time to time.

GunSmoker, 2020-06-12:
The problem is not sensitivity of power panic. 

Heaters are turned off - which means something in printer has "detected" power loss (yes, detected incorrectly, but still). The question is - why does printer itself continues to operate normally?? 

This is the issue - inconsistency in power loss detection between two components. Either printer should detect power loss and stop/pause, or it should not detect power loss and continue to operate.

The current behaviour (heaters are off, steppers are on) is a complete mess. Prusa is not a heisenberg's cat.

leptun, 2020-06-12:
The heaters are turned off in hardware, not software.

GunSmoker, 2020-06-12:
@leptun Yes, so? Why software is not notified about this? This is the issue.

Kachidoki2807, 2020-06-12:
According to the Power Panic schematic, there are differences between the silver PSU and the Delta PSU. The optocoupler and mosfet as well as the RC time constant are differents.
I'd be curious to know if the delta variant has the same issue.

Regarding the issue itself, indeed the heaters are turned off by hardware, but the firmware doesn't react to the nAC_FAULT signal. Or at least it is not visible. I suspect some high frequency ripple to be visible by the hardware, but filtered by the software, pure assumption.

If I find time, I'll put a scope probe on this signal to understand what happen.

closed 2023-09-07 by github-actions[bot]