prusa3d/Prusa-Firmware #1163: MMU 2.0 load / unload issue

danielegger1, 2018-09-17:
Firmware version:
3.4.0 final

Tested:
Provided Gcodes for the MMU v2.0

Filaments used:
Prusa / fillamentum filaments

I have tested several times the provided Gcodes and have recognized several issues with the load / unload sequence which causes that I cannot even run one print without any issue. Most problematic is the missing layers.

Issue 1:
Finda detects filament but get`s stuck in the orange PTFE tube (the longer one to the extruder) which causes the filament to never reach the extruder and I get missing layers in the print 
The above mentioned issue can happen for several reasons, the main problem is, that the filament sensor on the extruder is not used and this leads to an not detected loading issue

Solution:
Optional enable the filament sensor, for most of the materials it works as it is supposed to work only for some filaments this cannot be used, but without the filament sensor it will be never stable!

Issue 2:
Sometimes it can happen because of stringing on the filament that the finda detects too early the filament and the filament does not reach the extruder far enough, which causes missing layers.

Solution:
Move the extruder gears synchron with the MMU gear to load to filament farer which can reduce the faults by a significant factor.

Issue 3:
The filament cannot be unloaded from the extruder (because of a small clog which could be fixed with some hand force when the extruder is hot). The problem is after the messagebox appears on the printer the extruder temperatur is switched off. When I try to unload the filament now, I cannot get it out of the extruder because preheat is not possible. The MMU message needs the possibility to preheat before trying to fix the issue otherwise there is only a cold pull possible.

Solution:
Preheat possibility at the MMU message.

Already tested and checked:
- MMU idler screws force (tested different forces)
- Extruder idler screws force (tested different forces)
- Print temperatur changed +-5° 
- Center of the gears (the filament passes perfectly in the middle)
- Checked the assembly of the MMU (bearings etc. moves freely)

This issues mentioned above are not only detected from me, there are several other people with the same / similar issues and we cannot use the MMU as it is supposed at the moment.

[image]
[image]
[image]
[image]
[image]

gblairza, 2018-09-18:
I am experiencing the same issues

drmaettu, 2018-09-18:
I have the same problems. The suggested solutions sounds good.

mylife4aiurr, 2018-09-18:
I had alot of issues when I first got my MMU2. Most were stringing issues. I have greatly reduced stringing by lowering print temps and increasing ramming settings. I also calibrated my bowden length, which has also helped (please add bowden length calibration to setup guide)
 
One of the problems that remain is MMU2 + MK3 is unable to detect a lack of filament in MK3 extruder. It seems the like the mmu2 can only detect that filament is loaded into mmu unit (finda). But it cant detect **a lack of filament moving/being feed in mk3 extruder**.......this leads to entire colors/filament missing in mmu prints. Just because filament feeds from mmu doesnt mean the extruder has filament..... 

Is the filament sensor in mk3 extruder disabled, when mmu is in use??

darrellcfol, 2018-09-19:
Same issues here. Latest from support was to re-flash the firmware, but I have already done this 4 times already. The MMU is no where near the state it needs to be in for me to feel comfortable leaving it alone, and expecting a print to finish with no issues!

Has a very beta testing feel to it.

mylife4aiurr, 2018-09-19:
Slice it and forget it index -
MK3: - A to A+
MK3 with MMU2:  F to D-

I'll try and help you. But I'm still getting the issue I mentioned above myself.

1st step calibrate your bowden length.
Use filament prusa has tested. Prusa, hatchbox, etc - any filament in MMU2 profile.
If you are still getting stringing with prusa tested filaments, reduce temps....
If that doesn't work, increase ramming settings also.
For Inland pla which isn't a Prusa tested filament I had to lower temps and increase ramming. That's reduced stringing, but I'm still having failures because of problem mentioned above....

drmaettu, 2018-09-19:
Yeah, i love the mk3. Its a wonderful machine. But the MMU2 is just not ready. At least it isnt for me. 
What and where is ramming?

danielegger1, 2018-09-19:
I truly understand prusa as the first units which goes to new users will bring up new issues. I am sure that the bugs will be fixed soon. The most important thing from development point of view is, that they will get as much details as possible, so it will be easier for them to analyze and develope it. I am also disappointed about my MMU at the moment, as it does not work as it is supposed to work, but I am also sure that they will release a new firmware soon that we can get our prints out of the MMU as they are supposed to be.

Please all of you if you see posts about MMU problems let them know to insert some comments here.
From my side I have hopefully explained as much as I can but please guys put in some more picture, details, informations about your problem for the developers.

mylife4aiurr, 2018-09-19:
> Yeah, i love the mk3. Its a wonderful machine. But the MMU2 is just not ready. At least it isnt for me.
> What and where is ramming?

There is a better, more accurate explanation in slic3er of what ramming is/does.
Pulling hot filament from extruder and then moving it back and forth thru bondtech gears to shape it better for later reinsertation when that color/filament is needed again. If ramming isn't done filament tip may be to big or to stringy to go back into extruder later. But ramming adds to total print time. Also this is going to be very very filament specific. 

Ramming settings are active if mmu printer profile is active. Then goto filament tab, then advanced.

mylife4aiurr, 2018-09-19:
[image]

MK3 + MMU2 has no idea the green isnt feeding into extruder. As you can see in the photo the yellow is higher then the green. So yellow is feeding correctly, green is not. But finda says there's green filament. so it will continue with print. 
Eventually it may self correct but thats a maybe. Layers skipped = Print is trashed. And I have to babysit printer because it cant catch this condition.

darrellcfol, 2018-09-20:
Has anyone here experienced the retraction difference on occasion? 

What I have seen (with the same tension on the MMU idler door) sometimes the MMU will retract the filament where it’s tucked in the body and sometimes the filament is sticking out of the body just a little. 

I’ve had a busy week and shelved it, but I will try the ramming in Slic3r tomorrow. Really not looking for more failed prints though because it’s very frustrating. 

Glad I stumbled upon this because I don’t feel alone anymore though I had a feeling since they have been shipping for weeks and I haven’t seen many people bragging and showing off prints. If the “thousands” that have shipped surely I should have seen more posting completed prints.

mylife4aiurr, 2018-09-20:
@PavelSindler @XPila 

This is another attempt at printing this fidget cube, requested by my son. This time I got farther - yes. The point however is the mk3 + mmu2 cant detect when filament is present in mmu but not mk3 extruder. This leads to missing layers. On this particular print it happen in the green and yellow. 

Especially if end user isn't the best at building or tweaking despite excellent instructions and build videos, the mk3 + mmu2 needs to be able to catch this condition. 
 
[image]
[image]

darrellcfol, 2018-09-20:
I know the feeling... My daughter wants an Om Nom and my son wants a Batman
[image]

stahlfabrik, 2018-09-20:
That is brutal. I am so sorry that you are struggling so hard to fulfill the kid's wishes:-(

mylife4aiurr, 2018-09-20:
I think I downloaded all those same models from thingiverse...... hope a fix is released soon.

At 1st I thought I was the only 1 having issues. Your failures make me sad for us all

darrellcfol, 2018-09-20:
I was told to email them with videos of what was happening and after days of waiting all I got back was to re-flash the firmware which I have already done multiple times.

I also hope a fix is in place soon!

PavelSindler, 2018-09-20:
Thank you for these information. We will look on it closer.

danielegger1, 2018-09-20:
Looks like we need to buy some Prusament urgently :D

darrellcfol, 2018-09-20:
Just placed an order

drmaettu, 2018-09-20:
So i had one single print that wasnt failing in each day, several hours, lots of filament, huge amount of swearing, fiddling, taking it apart, reassemble. I think, for now im done with it. I don't have the patience, willing, energy anymore for one failure after another. I will put it aside, until prusa can solve this problems

darrellcfol, 2018-09-20:
Trying the Om Nom again now. 

1)Tweaked the ramming to 3.5 and print temp from 205 to 200.
2)Re-flashed the MK3 and MMU boards. 
3)Set the Bowden length again. 
4)Cold pull on nozzle 
5)First Layer calibration
6)Cut New tips
7)Load all for Filament 

Filament sensor is on, but I have not had a layer change yet. Will post an update later!

darrellcfol, 2018-09-21:
Failed to load on one layer, then it was able to load on the next layer. Filament sensor didn’t care. 
[image]

murphy2712, 2018-09-21:
Chris Warkocki shared some useful tips to get good tips 😉
https://www.youtube.com/watch?v=v1h-S7bxPEE

Blanchard3d, 2018-09-21:
Issues also, I have good points by messing with ramming and retraction but the issue I'm having is the "bowden length" programming. Since you can only set it with number 1 I have to set it to skip 3 times to make #2,#3 and #4 to work properly, they will barely grab. Then #5 makes 3 skips also. I don't know if somehow vibration makes the filament slide back when another one is being used or when positioning. I aligned everything with a 2 mm gage pin and checked tensions over and over, alignments are good and smooth. I tried 3 ways of setting the spools, all side by side, spaced out and elevated with the same exact results. I twisted the PTFE tubes made them straight as possible, put a loom on them a few inches down from the MMU with same results. I also notice that the "bowden length" changes. It will start with the #1 3 skips but about an hour in it will not be skipping anymore and then the other 3 filaments will not hand off which means you are printing air because there is no sensor at that point. I would rather have an option to turn the filament sensor on or off or placing a FINDA at the extruder might be another solution. I can get through a test I do that goes through all 5 colors in about 22 minutes but once you go to a real print something always goes wrong :( 
[image]

darrellcfol, 2018-09-21:
Great video linked above. It helped me with two areas of friction that was causing the main issues. 

Still think that a proper “handshake” between the FINDA and the filament sensor at the extruder is a must and it would practically make it indestructible.

Slight bleed because I didn’t mess with the purge out length from blue, but we have a happy dad and very happy little girl in the house currently!!!
[image]
[image]

darrellcfol, 2018-09-21:
@Blanchard3d looks like your FINDA is lower than mine in the picture but it could be the angle of the picture. I’ll check mine when I get back home.

Blanchard3d, 2018-09-22:
@darrellcfol I used my tips to set the ball now that I have consistent good ones. I also left a little clearance in case it flares a little bit. 
[image]

Blanchard3d, 2018-09-22:
I think the key is to get your ramming points good and then adjust the FINDA to drop off the filament where you want it. You can then calibrate the length from the drop off to the extruder in the service menu. It works but after awhile it seems to reset the calibrated length back to default or the driver is re-zeroing somehow. I look at the filament dropoffs and they are still the same as before but I have to add a bunch of steps to the calibration length again. It will work for awhile then the same thing happens, again the dropoff locations are still where I had them. I even flashed the MMU a couple of times but same thing.
Is anyone else seeing it load the correct filament number then unload it, move to the front and up in Z and load the wrong filament, purge in the air and then print with it like it was the correct filament number?

DavaD85, 2018-09-22:
Same problem for me, seems the calibration reset sometimes...
Tried flash different time.

drmaettu, 2018-09-23:
[image]
Yesterday i had, finally, my first successful print. After that, i started a new print, sliced with the exact same settings, printed with the exact same filaments on the exact same setup.
[image]
This is so sad. And a waste of filament. Please Prusa, fix this

Blanchard3d, 2018-09-24:
Must be something in the firmware that tells it to kick out a extruder # if it doesn't like something about it and go to the next extruder # and use it instead? 
All the white in this print was supposed to be red. 
It loads the red and I hear it reach the extruder gear but immediately it lifts up in Z, moves to the front of the bed and unloads it. Then it goes to extruder #1 (white) and loads it, extrudes in the air a bit then continues to the purge block and prints as normal.
It also lost the gold (extruder #3) the last few layers in this print that was aborted but when I open the MMU the filament is sitting in the correct place and the point looked OK. 
I will reflash and check alignments, tensions, ramming and FINDA again. I have only been printing with the SD card also. 
[image]
[image]

mylife4aiurr, 2018-09-24:
I got a little farther than you @Blanchard3d. I thought I had my filament tips "ok - not perfect". But "ok" is good enough for a 15 hour print. Orange solutech3d and true black hatchbox have ramming settings that are tweaked good enough to not string this entire print. The white almost made it. I have to work on the white solutech3d profile.

[image]

I will not give up. I think I'm really learning to optimize this mmu2. 

However excepting the user to use only Prusa tested filament that has good ramming settings with mmu2 isnt reasonable. On the same token hoping the end user has a perfectly tuned, resistance free filament path to avoid jams and layer skips is no substitute for actively detecting if filament is in the extruder. I mean there is a laser filament sensor in extruder...

danielegger1, 2018-09-24:
I see there are a few values what are possible to change can you explain which obe did you modified?

Side note:
I am using prusa PLA and it also does not work with the provided settings

Meanwhile I can eliminate mechanical issues as most of the times a print failes because of strings which blocks loading / unloading.

darrellcfol, 2018-09-24:
I got my first 5 color print to finally go the entire way. It took 20 hours but was only supposed to be 14. Reason it took so long was I had the temp to high on the eSun filament and the string was long enough to keep the ball bearing from dropping from the FINDA so the Hobbes gears in the MMU2 just ran the filament our the back so it didn’t engage on the next round. 

That error is easy to recover from and I fixed it the next morning when I woke up. I think I have the eSun PLA+ printing just fine now. 

I think going forward I’m going to find a small common model I can test the filaments when buying new filament colors or even brands, so that I can check the tips and temps to dial it in before starting a bigger print. 

https://www.youtube.com/watch?v=po-9-X91ZfY

darrellcfol, 2018-09-24:
I did loose one today because of power loss, but wasn’t home to see exactly what happened but the parts that were white had started to print as layers of blue. Kind of like the selector didn’t know where it was in the print. 

Since I wasn’t home I don’t know what it did, but I might just trip the power manually on a test print to see what happens and if I can replicate it I will report it to the Prusa team.

Blanchard3d, 2018-09-25:
@danielegger1 Try different retraction speeds to get your points good. I made a single line 10mm long and imported it 5 times, gave each one a different color and sliced it. This lets it prime and make the point on each filament and takes about 10 minutes to run through the program for all 5 filaments. I just set each filament up 5 higher on retraction from the next until I saw good points and got them close that way. They are not perfect every time but mostly are good. The small PTFE tube on the extruder has a smaller ID and a string folded over might not pass through.

I did bodwen length calibration tests eliminating the spool holders and the spools of filament themselves, eventually I also eliminated the PTFE tubes in the back. Since the printer does not know if it has material in it or not (why I have a sensor there?) I used that to my advantage and unscrewed the bowden tube connector on the extruder to use as my measuring 0 point. Test results with no bowden tubes on the back and bowden tubes are the same, test results with the spool holders attached vs just the bowden tubes on the back vary.

I decided to get rid of the spool holders and hang the spools in the air. Bowden length calibrations come out the same every time I measure them this way. So now that I feel the friction problem is solved.

I cut the ends flush with the bodwen connector after the MMU finished pushing them and used them as my measurement to verify each test to make sure the length was not changing. I've repeated this test over and over with the same exact results. 

Resulting test conclude that #1 and #5 are much longer than #2,#3 and #4.  the difference in my case is 50mm! This means it will go into the extruder for sure but will also grind your filament, will cause more problems down the road. The results are in the picture. 

To sum it up if you are having varying bowden calibration lengths and have lined everything up on the MMU (I used a 2mm drill bit) and everything is smooth in the unit to the extruder you might want to check the spool holders. 

The difference on #1 and #5 compared to the rest is 50mm, it seems to be in the firmware. On purpose?
[image]

bozzzzo, 2018-09-25:
@Blanchard3d measure your hobbed gears in the MMU, mine had quite a discrepancy in diameters, which leads to different feed lengths among positions.

AbeFM, 2018-09-26:
I had this as well, swapping who's on first halved my error, but I still can fall short by over a CM when feeding.

Blanchard3d, 2018-09-27:
@bozzzzo this is true, I measured the OD of the hobbs and 3 are the same 2 are oversized by .016" this sixteen of an inch translates to 50 mm over the entire length of the bowden length. I rearranged them to be sure and it follows the hobbed gear.

FasterThanLlamas, 2018-09-28:
+1  I have extended the feed until the left button no longer responds and am still short of the extruder.

mylife4aiurr, 2018-10-02:
Guys I found today. We should be posting mmu2 specific firmware issues:
https://github.com/prusa3d/MM-control-01/issues

AbeFM, 2018-10-02:
""They are working on individual pulley calibration, idk when it will be ready. but good news""

How do you know this?

AbeFM, 2018-10-02:
Can someone tell me how to go from that link to flashed firmware?? I cloned the repository, but the target processor isn't the same (32U4 vs 328P) so I'm a little unclear if I should plug my MMU into the laptop and hit "compile" in Arduino.

mylife4aiurr, 2018-10-02:
1st error.
let it create a folder "MM-control-01" then move all files from zip into that folder

2. compiling
from MM-control-01.ino
_// MM-control-01
// HW arduino Leonardo_

For me that complied no issues.... no I didnt try to flash it... not that brave

AbeFM, 2018-10-02:
I renamed the folder the zip was in. Probably it should be named that before it is zipped, then the unzipping program will more likely default to the right one.

I had no issues compiling under Leo.

FasterThanLlamas, 2018-10-07:
> Wow things are a lot better with the new firmware.
>
> It seems like it handles jams better, too - reloading filament through a filament change as I used to do manually (after "finishing movements").
>
> I compiled it as shown above. I had a lot of issues getting the com port to work right - hitting reset at JUST the right time... But in the end it worked and I've got all the lengths calibrated!

When do you think it will be ready for distribution?

darrellcfol, 2018-10-07:
> Wow things are a lot better with the new firmware.
>
> It seems like it handles jams better, too - reloading filament through a filament change as I used to do manually (after "finishing movements"). [...]

Awesome. I have my smaller gears on 1&2 already. I’ll wait for them to post the easy flash because I’m sure I would screw something up trying to monkey with it. 😆

AbeFM, 2018-10-10:
I take it back. I've had to abandon a bunch of prints because the selector and the drum are rarely in the same place.  When the MMU needs work,  you can never use the buttons to get it into a state the printer wants it in.

We need to be able to load and unload filament, not just when the printer tells the MMU to - or else the printer needs to be a LOT more aware of what the MMU is doing.

Similarly, homing the MMU anytime things aren't 100% as expected should be done - it gets off a high percentage of the time, and there's just not harm in doing it.

Explicitly rehome and put the MMU in the state it needs to be in (move the selector and drum).

I'm not sure how to write this up as a topic, but it's a big one. I'm staring at the printer, it says "loading filament 2", before the drum was on 3, the selctor on 5, the light on 4. After a reset, I can move the selector to the right spot, but I can't move the process forwards at all, just move the selector around and "preload" the filament.

 But it would be GREAT if I could tell either the printer or the MMU to gets it state from the other device

Blanchard3d, 2018-10-10:
@AbeFM I see this issue also. 

I don't have this issue a lot now, but when you do it sucks, especially because the print times are so high and wasted filament. One thing I do before each print is clean out the MMU and even out the tension in the springs when putting the cover back on.  

I can get good prints now that I have the same diameter hobbed gears but it seems the rotating drum and selector only knows home position. 
When the drum gets out of sync you can take off the top 2 screws, open the MMU and rotate the drum where it will match the hobbed gear when closed. There is no resistance to hold the drum in position and moves too easily. 

The selector will not move though because the stepper is activated, so only a reset can help. Once you reset the machine will happily go about printing air and MMU will be on filament 1 even though the printer thinks its on a different filament station. 

The other problem I see is with the MMU is in the help me state the hotend is off and the extruder motor is activated. There needs to be a way to heat up the hotend and move the E axis.

AbeFM, 2018-10-10:
I move the drum (thankfully I didn't get dad's fat fingers) with it all closed up, I can either reach in from the rear-left or front right.

You can manhandle selector the stepper, but it's too hard to know if you're in just the right spot. Sort of the same with the drum.

Mostly I reset it over and over and over until it happens into a mode where I can move the selector. Most of the time, this just gets the printer printing with an empty extruder, but once it's doing that I can get to the menus, select "change filament", let it leave a half a missing layer (hopefully in the purge tower) then there's a 70% chance I can get it to feed.

I believe the MMU needs a screen, or the printer needs a massive amount more control over the MMU, so we can control it from something with contextual feedback. A light that blinks after a light that was solid is hard to remember/interpret, and even then you can't actually instruct the machine to do something useful.

It feels like a premature baby. Once another 50 pages of code gets written (error handling, manual control, resting steppers, actually using current sensing, using filament sensors... OK, a few hundred pages of code), it'll be pretty sweet.

JohnSvensk, 2018-10-14:
> Looks like we need to buy some Prusament urgently :D

Hmmmm... *conspiratorial thoughts* ;)

BlindSniper35, 2018-10-24:
After adjusting the five different filament lengths With the mmu 2.0 firmware update. The feeding issues drop somewhat. I'm still get feed issues that cause the filament to never reach the extruder. Sometimes this is caused by the tip having a bit of thin wisp string that folds over in jams the orange Bowden tube. Other times I'm just not sure why it did not feed enough. 

On a print there always be at least one failure to load that causes a layer skip(average on the lizard it's about five or six). Generally the filament that cause the layer skip the next time it is fed into the extruder as a pretty good chance that it will load the next time and print correctly except for the fact that were there is no layer underneath.

It seems like the best solution would be to enable the filament sensor on the Mark 3 extruder. If the filament sensor is causing too many problems or can't be use it due to the filament properties. It might be worth adding a menu option to try to reload the filament and start that colors section of the print again. So the load failure and layers skip if caught can be recovered from.

AbeFM, 2018-10-26:
I posted some weeks ago a simple fix for the hub - trying to hold large torques with a few mm level arm is ridiculous.

[image]

Adding a cut out like this lets you apply large forces with weak materials - this is a TPU knob that lets me feed filament into my CR-10 through a long bowden tube and it NEVER slips. A gear is a matter of pennies and could fix that issue.

BlindSniper35, 2018-10-28:
I also had pieces cracked due to them not being strong enough or holes being under sized(most of the cracks were in press-fit pieces for bearings). The pieces that cracked I replaced with higher number of perimeter(also use calipers to double check dimensional accuracy). That seem to solve a  most of of the inconsistency I was having. that did not fix failing to feed enough filament sometimes. It did fix feeding the wrong filament and repeated jamming due to misalignment.

AbeFM, 2018-10-29:
It's just not a great design.
"How much do I turn the screw"
*I can't tell you
"What happens if it's too loose"
*The printer won't work occasionally
"What happens if it's too tight"
*You break the printer and have to make new parts

It just doesn't seem like a good approach if you can accomplish reliability with a small/simple/cheap change. Sometimes you're backed into a design corner, but not here.

dancorrigan1, 2018-11-02:
Issue 1:
Finda detects filament but get`s stuck in the orange PTFE tube (the longer one to the extruder) which causes the filament to never reach the extruder and I get missing layers in the print....

Check. I've had that happen multiple times.

Issue 2:
Sometimes it can happen because of stringing on the filament that the finda detects too early the filament and the filament does not reach the extruder far enough, which causes missing layers....

Check. I've had that happen multiple times.

Issue 3:
The filament cannot be unloaded from the extruder (because of a small clog which could be fixed with some hand force when the extruder is hot). The problem is after the messagebox appears on the printer the extruder temperatur is switched off. When I try to unload the filament now, I cannot get it out of the extruder because preheat is not possible...

Check. I've had that happen multiple times. ... Wow.. This one is pretty stupid too.  Turning off the heater but requiring it to be on to the fix issue, yet giving no way to turn it on!!!

johnnycope, 2018-11-02:
The problem with the FINDA probe seems to be pretty common, however it is also relatively easy to fix. The FINDA adjustment should be just touching the filamen, there should really be no resistance between the filament and the ball, the ramming setting can also be tweaked to get a nice point. When setup right it is mostly trouble free , the heater will switch off but does give you a minute or so to give a bit of time to clear the error (possibly not long enough) . There is still a bit of tweaking to do with this design and implementation. But my one works absolutely fine.

dancorrigan1, 2018-11-02:
When I walk away for 3 hours..like I do every single day with normal
prints, The printer will have already cooled down the hotend by that time.
If I have to babysit the entire print to watch out for this behavior.. this
defeats the purpose of having the MMU.

johnnycope, 2018-11-02:
The hot end shutdown is a standard feature of the original firmware and will turn off if the printer is idle regardless. This feature could be disabled in the firmware as could enabling the filament sensor but doing so would disable an important safety feature. 
I think it is more important to stop it happening in the first place . For this I the ramming and retraction settings can be adjusted to do the job.

dancorrigan1, 2018-11-02:
How about an option to reheat when I’m ready to work on it, rather than locking up the LCD/knob 

Thanks,
Dan

B3D-P, 2018-11-02:
Not sure about the safety protocol, my extruder melted because of the filament sensor while running a multicolor print on the mmu2 mk3.

[77 more replies]

closed 2022-07-21 by Prusa-Support