prusa3d/Prusa-Firmware #1164: Purge Line  Different Lenghts

christizzz, 2018-09-17:
Hello,

The Purge line are getting different lenghts randomly... the start code is the default one.. sometime it is the short default purge line.. and some times is 160mm long... it seems that the printer has some issue between relative and absolute for this purge line... Relfashing didnt change anything.. it is not so problematic.. but the issue is therr

stahlfabrik, 2018-09-17:
Yep it has been like that since I got my MK3 last year. Sometimes it is double length

Moorviper, 2018-09-18:
do you use different materials then it would make sense

Panayiotis-git, 2018-09-19:
IIRC, I had the same problem sometime ago with the MK2.

You can add the g-code G90 to the start g-code of your printer profile, just after mesh bed leveling.

It will switch the printer to absolute positioning mode.

Area5142, 2018-09-19:
From time to time, I have seen this randomly longer purge line since I got the MK2.
It is also mentioned in this thread on the forum:
https://shop.prusa3d.com/forum/general-discussion-announcements-and-releases-f61/print-keep-stopping-to-unload-and-load-filament-t23245.html#p99172

GWdd, 2018-12-27:
I have the had the issue since new, never been a problem.  But, caught a file that does it repeatedly; every time.  The symptom is the volume is stuck at the first section purge rate, so the second section requires more length to extrude the volume.  So I looked and sure enough the file also has the MaxFlow set down from 15mm3/s to 10mm3/s.   I'll test if this setting has any effect and add to this post.

AbeFM, 2018-12-30:
How did it go? I've noticed this as well.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

wavexx, 2019-08-23:
Also #1494

closed 2023-09-07 by github-actions[bot]