prusa3d/Prusa-Firmware #1170: Change the direction of rotation of the controller in Main Menue

Devilscave, 2018-09-18:
It bothers me since I got my MK3 and I just can not get used to it. The direction of rotation of the controller to the menu is the wrong way around. Some things but like temperature regulation are right around. If I turn clockwise it goes down and around. For this purpose, the controller would have to sit to the left of the display. I have now resolved this error myself by crossing the two traces on the control board. Now, however, the settings of the individual values are the wrong way round, but I have to run there much less often than in the menu scrolling back and forth.

Devilscave, 2018-09-29:
I took care of the problem myself. You can find instructions on Thingivers.
https://www.thingiverse.com/thing:3103099

cimoalpacino, 2018-09-30:
If I'm not wrong the direction could be inverted in firmware.

Devilscave, 2018-10-09:
That's why I opened this message.

cimoalpacino, 2018-10-11:
No need to solder.
Everything can be done in FW. 

https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/lcd.h

Just need to play wit encoder settings and thats it. 
See original Marlin FW examples and for different types of boards.

Devilscave, 2018-10-12:
OK, and how can I load this part of the code into my firmware if it is always specified as a coded hex file?

Devilscave, 2018-10-12:
Should go in the firmware if you change the values x0B & x01 in line 214 & 215. But how can I import this?

pedjas, 2020-05-04:
This is really annoying issue and as it looks like easy to fix in firmware
Hope it will be done soon.

closed 2020-05-04 by Devilscave

Panayiotis-git, 2020-05-04:
@pedjas If you have the original Prusa display or a compatible one (see LDO) everything works fine.
If you have a RepRap display then you have to modify the FW yourself or do the hardware fix.

pedjas, 2020-05-26:
I have original display controller. No it does not work fine. 

Menus rotates are backwards. But some other settings are rotated normally.

Those are two issues: one that rotation is backward (by some opinions which is debatable) and second: it behaves differently in different parts of interface (which is not good at all).

I could follow advices to change wiring or move button, but then I would fix one issue and then other parts of interface would be moved backward.