prusa3d/Prusa-Firmware #1172: 3.4.0 Firmware - LCD font broke after print

cerw, 2018-09-19:
Hi

Flashed my i3 Mk3 and after 12h print I saw this on my LCD screen: 

[image]

Did not see any odd behaver just thought it would be useful to report.

Thanks for making great printers!

PavelSindler, 2018-09-21:
Hi, thank you for reporting this.

rdavolio, 2018-09-22:
Same issue with me

cerw, 2018-10-05:
This is still issue even on 3.4.1 only after successful print.

PavelSindler, 2018-11-26:
Does this appear after every succesfull print or is does it appear randomly? This may be caused by static electricity or broken LCD cable/connection.

cerw, 2018-11-26:
After a successful print.

PavelSindler, 2018-11-26:
Each time?

cerw, 2018-11-26:
_Ano prosim_  Yes