prusa3d/Prusa-Firmware #1173: Z shows ?

Debiano, 2018-09-19:
Hello,

with the update to 3.4.0 Z shows always "?" for me. If i do the Live Z adjustment it shows "?" too, even if i change the value. Its just if i compile the firmware by myself and just with the last 3.4.0. If i use the pre-compiled file it works.

I am the only one who have this issue?

Best regards

Panayiotis-git, 2018-09-19:
No you are not the only one. I've been there and a lot of others who skipped some steps from the compile instructions at Prusa's github page. Especially, the paragraph 1.3.

Check also the #1117

3d-gussner, 2018-09-19:
And never ever update the AVR boards when you start Ardunio IDE 1.6.9 it will generate errors. Use only Arduino AVR Boards version 1.6.11.
I have Arduino IDE 1.6.9  in portable mode to separate it from other Arduino instances:
https://www.arduino.cc/en/Guide/PortableIDE
And for this i changed the Sketchbook location <your path>..\arduino-1.6.9\portable\sketchbook\libraries

Debiano, 2018-09-19:
Thank you very much.
The method with the portable version helped me a lot and it works now.
I just wondered, because in earlier firmware versions it worked without problems and i changed nothing. Maybe i just had luck.

I wish you guys a nice day! And btw. @3d-gussner thank you for your work with the hyperfine bed leveling! :)

closed 2018-09-19 by Debiano