prusa3d/Prusa-Firmware #1178: "filament sensor response is poor, disable it?"

geeksville, 2018-09-19:
Hi,

I just upgraded from 3.3.1 to 3.4.0 (final).  I'd like to report a regression:

1) Previously the only time I had a problem with the filament sensor false positives was transparent red filaments.  I could print all day long with transparent and opaque filaments of various other colors.

2) Since upgrading to 3.4.0 a couple of days ago I've seen this warning for approximately 50% of the filaments I've tried.  These are not exotic colors - just white, yellow and blue (some PLA some PETG).

It seems to me your new sensor engine might be a bit too selective.  I'm happy to capture whatever debugging information you'd find useful (I'm an embedded sw eng for my day job and can build your fork of Marlin from source if you'd like any particular debug info added).

I'm considering downgrading to 3.3.1 or just ignoring that warning...

Keep doing such great work!  Bravo!

kcnjamie, 2018-09-20:
I am having the same issue since upgrading.  Colors light blue, green and red have all given me this error.  These did not cause an issue with the prior firmware.

SherifNagy, 2018-09-20:
I had the same issue with PET-G Prusa's orange filament

PavelSindler, 2018-09-20:
Yes it seems that new warning "filament sensor response is poor, disable it?" is too sensitive unfortunately. Did you see this warning usually when loading filament before starting print or did you see it usually when filament runout detection triggers and filament change is performed?

jettoblack, 2018-09-20:
@PavelSindler I'm getting this too and I see it in both cases, but more often when loading filament. I've also noticed that filaments that worked fine with the sensor on in 3.3.1 are now frequently triggering the behavior during a print where the print pauses and the filament is retracted and re-inserted a few times to check if it's still there.

kcnjamie, 2018-09-20:
I have only experienced this during filiment load.

geeksville, 2018-09-20:
@PavelSindler I saw it only when loading the filament, and then I said yes to disable the sensor (so it wasn't on for later portions of the print - and I haven't run out of filament yet anyways)

painnovation, 2018-09-20:
I can report this happening with Hatchbox orange PLA upon load of filament. Never had a problem with the sensor previously before upgrading.

cmscience, 2018-09-21:
SAME! So frustrated Paramount 3D PLA Ivory can get past the first layer and if I select yes disable printer it craps out and drags the nozzle through my first layer runing the print going back to 3.3.1 until my mmu 2.0 gets here in 2019

0not, 2018-09-21:
I am experiencing the same issue with white PLA from Inland. I was in the middle of a small print and the filament sensor beeped and asked me to pull out the filament. I followed the instruction to replace the filament and ran into this same "sensor response is poor" error. [What follows might be another problem that deserves its own issue.] After disabling the sensor, the printer tried to resume the print. The head moved back into position, but then just sat there. I couldn't control the printer via Pronterface until I restarted the printer.

bsrotten, 2018-09-21:
> I am experiencing the same issue with white PLA from Inland. I was in the middle of a small print and the filament sensor beeped and asked me to pull out the filament. I followed the instruction to [...]

This.  Mine did the exact same thing 4 hours into a print.

PavelSindler, 2018-09-21:
Thank you all! We have already fixed issue with resuming after filament change in PR https://github.com/prusa3d/Prusa-Firmware/pull/1174. We will probably disable "filament sensor response is poor, disable it?" feature for now and release 3.4.1 as soon as possible.

dikodahan, 2018-09-21:
> This. Mine did the exact same thing 4 hours into a print.

+1 to white PLA and PETG. 
With previous FW I had issues with black filaments and now that seems to be resolved with 3.4.0, but with white I am having worse issues than previous FW with black.

jhron, 2018-09-22:
+1 to white Prusa PETG
Every time when I load white Prusa PETG filament I get message "filament sensor response is poor, disable it" (I did not try another filament yet). 
I did several successful short prints (I have only noticed that sometimes printer stopped moving head for a while - around 1s, but I am not sure if it is related to this problem).
Then I did 2 longer prints were I was forced to unload and load filament randomly.
First took 5h 34m and I was forced to unload and load filament 3x during this printing. When there was question if I want to disable sensor, I selected no every time. In this case printer was able to continue where it stopped every time. So on the end printout was finished successfully (only problem was that it was during night and I was woke up by printer beeping).
Second print took 4h 13m but it should take around 8 hours. I was just woke up by printer because of it. I was asked to unload and load filament - so I did it and in additional, when there was message "filament sensor response.... do you want to disable sensor? I pressed yes. After that printer head moved back where it stopped before but this time it did not continue. Printing was interrupted.
I went to Octoprint and there was message that there was too many timeouts. I don't know how long was printer beeping in first and second case until I woke up and replaced the filament and why in first case printer was able to continue and in second it was not.

My filament sensor is perfectly clean - I am sure because I did upgrade to latest printer head revision and I cleaned the sensor during upgrade process.

marcopiraccini, 2018-09-22:
Same here with white PLA. 
Loading the filament: "filament sensor response is poor, disable it"
...and during the print, after approx 5 minutes, the firmware asked me to change the filament.
I had to disable the filament sensor to complete a print.

rkayakr, 2018-09-23:
Also lost a print due to filament sensor failure with orange PETG and NO resume after reload. I went back to previews release. Please fix this.

kcnjamie, 2018-09-23:
@rkayakr if you read all the thread you will see they are already fixing it.

geeksville, 2018-09-23:
Hi @PavelSindler sounds good, though I'm a bit worried that just removing the "filament sensor response" precheck will be sufficient.  It seems like one of the goals of that precheck is to prevent the pauses during the print that people have been reporting on the forum (#1186).

rkayakr, 2018-09-23:
I saw it. I look forward to a revision that is more reliable. In the
meantime I'll stick with 3.3.1 until my MMU2 arrives and I need to move to
3.4.x.

bdmihai, 2018-09-24:
hi... same problem here. The printer paused in the middle of the print due to wrongly detected missing filament (orange PETG). After reloading the filament the resume print did not work properly.

I also noticed that the printer pauses for a little time 0.5s during the print when it should not. Maybe it waits for the filament sensor detection...

jurriaan, 2018-09-24:
@bdmihai see #1187 for an issue about the stuttering problem

ThaliaFromPrussia, 2018-09-24:
Same here with silver PLA Prusa filament. Stuttering and asking for filament load, then not resuming or on false position.

bolsoncerrado, 2018-09-24:
I'm on the very same boat!

Just upgraded a MK2S to MkS2.5, and after the very first print with an all time used PLA filament we have in our farm (never had issues before), we got suttering (aka pauses every now and then where the printer seemed totally dead, to just continue moments after) and a final "sensor signal poor" warning.... Happened with RED PLA.

EDIT: We just upgraded a 2.5 to 3.4.0 and had the same issue with white PLA too. Different brand this time.

Crap!

bolsoncerrado, 2018-09-24:
@PavelSindler just note 2.5 is affected too. Thanks.

JohnOCFII, 2018-09-24:
Just as a counterpoint, I built a MK3 two weeks ago, and upgraded a MK2 to MK2.5 this past week.  Using black PETG or Prusa silver, I have not had this issue across approximately 40 hours of printing across the two printers.  So - it isn't impacting everyone.

bolsoncerrado, 2018-09-24:
@JohnOCFII  and you used 3.4.0 on both?

JohnOCFII, 2018-09-24:
> @JohnOCFII and you used 3.4.0 on both?

Correct.  3.4.0 on both the MK3 and MK2.5  Since they were both fresh builds/upgrade, I moved from 3.0.12 on the MK2 to 3.4.0 and, well, my initial prints on the MK3 were on 3.4.0.

JohnOCFII, 2018-09-24:
And just in case it is related, I'm not using OctoPrint, nor do I have anything connected to the serial port on my printer.  I'm not experiencing the issues others are reporting.  I see at least one person above was using OctoPrint.  Not sure if that is related, but wanted to put it out there.

bolsoncerrado, 2018-09-25:
I had both fails on one of each, aka 1 in SD printing with no USBs attached and 1 in Octoprint, so probably not related.

marcopiraccini, 2018-09-26:
> Just as a counterpoint, I built a MK3 two weeks ago, and upgraded a MK2 to MK2.5 this past week. Using black PETG or Prusa silver, I have not had this issue across approximately 40 hours of printing across the two printers. So - it isn't impacting everyone.

My guess is that it's somehow related with the filament color.

noradtux, 2018-09-26:
To me it seems to strongly depend on the surface of the filament (which I would expect). I have some very glossy filament which the sensor has issues detecting. On the other hand I just printed a largish object in transparent PLA which the sensor happily accepted.

bolsoncerrado, 2018-09-26:
Yeah but the thing is that they supposedly released an "engine 2.0" to avoid false positives and it turned out most if not all that the filaments we used to print FINE now are "rejected"....

koen330, 2018-09-28:
Same, same here. Unable to use the filament sensor after upgrading to 3.4.0. The "engine 2.0" is not doing it's job :(

creoPL, 2018-09-28:
I have the same problem , I have a green filamnet .

tonysmartantzoulatos, 2018-09-29:
Same problem here. Prusa Mk2.5 with latest firmware, on 99% of filaments i get the same message and i have completely disabled the sensor in order to print. No matter the color, or type of filament and so far i have tested in black, PLA, PETG, CFnylon, PC+, ZultraT & ABS and on PLA almost any color available, just to check. same problem each time!

marcopiraccini, 2018-09-30:
Tried the last 3.4.1. The "filament sensor responses" message is gone (make sense: has been disabled :)) but still the printer wants me to change the filament after 4-5 minutes of printing.
So back to 3.3.1 (which works perfectly...)

tedder, 2018-10-02:
Is there a M-code that will give the value from the filament sensor? I'm curious about the raw numbers.

TheFitzZZ, 2018-10-06:
Ran into the same issue today - was using Octoprint and it did not resume after returning to the proper position. Black PLA.

Side-question: does any1 know how to make Octoprint wait longer? The detection issue is bad enough, not being able to resume is even worse. :-(

shakti-baliyan, 2018-10-17:
After upgrading with new firmware few days ago, I had several successful prints although I saw poor sensor reading message whenever I reloaded a filament. 

Today I tried a new black glossy PLA filament which I had never used before and mid way through the print it stopped and asked for changing the filament. After changing the filament, it again gave poor sensor reading message. But I did not disable the sensor and continued. After a while same happened again and then again third time but this time I disabled the sensor. 

After doing this the extruder moved very slowly and then later sat on the same position for a while over the print. It appeared as if I have lost the print. However, after browsing menu I found Resume print option. On clicking resume print it resumed print normally where it left off.

bolsoncerrado, 2018-10-17:
Definetly buggy...

OutofDarts, 2018-10-27:
I run 22x prusa i3MK3 printers, and I'm experiencing wide-spread filament sensor issues. They both fail during prints, and when reloading. If a sensor gives me the "response is poor" message during a print, and I turn the sensor off- the print will fail 100% and the nozzle will crash into the print, as it resumes at the incorrect Z-height.

The majority of the printers have this issue, regardless of filament type, etc. I've printed twenty rolls of Prusament and it's happened on their filament too. The resume option NEVER works, and I lost dozens of prints before realizing it was the resume feature causing it (after disabling sensor).

I'm running 3.4.1. Very buggy software/hardware.

alexwhittemore, 2018-10-27:
> I'm running 3.4.1. Very buggy software/hardware.

Good to know it's not worth trying the upgrade again ;)

What's annoying is that people seem to have pretty good solutions to these problems, like indirect filament sensing via a bearing that touches the filament. I feel like it's not a huge stretch to get a Mk2.5/3 that actually does work generally quite well, but handling all the firmware tweaks off the main line makes that route super unattractive. I just want a standard printer that works, reliably.

shakti-baliyan, 2018-10-28:
>  I'm running 3.4.1. Very buggy software/hardware.

What issues are you facing in 3.4.1? Yesterday I switched to 3.4.1 and I am continuously running printer after that but have not encountered any issue even for the same filament that caused issues earlier.

bolsoncerrado, 2018-10-28:
Didn't you read his lengthy report? 3.4.1 seems hell for most but not all of us...

shakti-baliyan, 2018-10-28:
What report? Sorry, I can't seem to find it. Can you please point me to it? So far it works for me but still I am curious to know about issues in 3.4.1.

StevenWeathers, 2018-11-02:
Having the same issue, luckily wasn't a big print that it happened on but it did drag the nozzle across the print and scared the crap out of me.  I have since always selected "No" for the disable and its worked alright, but every filament load during a print has shown this issue of "poor response".  My other MK3 hasn't had this issue with the same filament but it wasn't upgraded to 3.4.1  yet.  Now I've upgraded it will see if it displays the same issues consistently.

3d-gussner, 2020-09-22:
@geeksville Is this still an issue with newest firmware? If not please close this issue.
[stale]

JohnnyDeer, 2020-09-29:
I´m closing this issue as stale.

closed 2020-09-29 by JohnnyDeer