prusa3d/Prusa-Firmware #1179: Printer MK3 pause mid print then resuming

SherifNagy, 2018-09-20:
Hello,

I just upgraded my MK3 firmware to 3.4.0 and changed my nozzle to hardened steel one, I was trying to print Raspberry PI Zero mount, I did slice the file using sli3r PE and added 5C to the default filament profile as suggested by Prusa's support due to that steel won't conduct heat as good as brass.

I am using Prusa's orange PET-G filament

While I am printing, the printer will pause mid printing for a second then prints again, while it is printing, the the icon of "SD" card on the LCD disappears.

I am not sure what is the cause of the issue.

Regards,
Sherif

PavelSindler, 2018-09-20:
Printer probably just checks if there is still filament. You could probably see extruder stepper movements during this "paused print".

PavelSindler, 2018-09-20:
Please if you will see this pause again, check if extruder stepper is moving and write here for confirmation.

SherifNagy, 2018-09-28:
it did happen again few times, just a pause the SD word gone and then printer keep going, I couldn't tell if the extruder stepper was working or not

SherifNagy, 2018-09-28:
I finally seen what is happening, at some point of the print, the extruder pushes away the filament a bit and then pulls it back in during this pause issue, here is a video, it actually was paused in the beginning of the video "sorry for using google drive"

https://drive.google.com/file/d/1AzYaDeiF4UsK2VUcPLOClk1Y2fLdzT5K/view?usp=drivesdk

closed 2023-09-11 by github-actions[bot]