prusa3d/Prusa-Firmware #1180: USB (Octoprint unusable)

jurriaan, 2018-09-20:
Since I couldn't get Octoprint to work because of all the timeouts / RX Buffer overflows (see #738 for some info about that) I was trying to use Octoprint while printing from SD.

But even that does not work. I seem to get weird responses from the printer resulting in disconnection eventually. While starting a print from SD the issues start like this:

<details>
 <summary>Start of disconnection issues</summary>
```js
Send: N159 M27*27
Recv: model_0.15mm_PET_MK3.gcode
Recv: SD printing byte 17762/661767
Recv: 00:06
Recv: ok
Send: N160 M105*32
Send: N161 M27*16
Recv: ok T:240.8 /240.0 B:86.1 /90.0 T0:240.8 /240.0 @:29 B@:127 P:52.5 A:41.8
Recv: model_0.15mm_PET_MK3.gcode
Recv: SD printing byte 18212/661767
Recv: 00:06
Recv: ok
Send: N162 M27*19
Recv: model_0.15mm_PET_MK3.gcode
Recv: SD printing byte 18572/661767
Recv: 00:06
Recv: ok
Send: N163 M27*18
Recv: fsensor_stop_and_save_print
Recv: fsensor_oq_meassure_start
Recv: echo:Enqueing to the front: "G1 E-3 F200"
Recv: echo:Enqueing to the front: "G1 E3 F200"
Send: N164 M27*21
Recv: fsensor_oq_meassure_stop,_prinmples
Recv:  st_sum=716 yd_sum=127 er_sum=0 er_max=0
Recv:  yd_min=43 yd1, Last Line: 162
Recv: Resend: 163
Recv: ok
Recv: or_err_cnt = 0
Send: N163 M27*18
Recv: fsensor_restore_print_and_continue
Recv: ok
Send: N164 M27*21
Recv: Error:Line Number is not Last Line NuN163 M27Last Line: 162
Recv: Resend: 163
Recv: ok
Send: N163 M27*18
Recv: model_0.15mm_PET_MK3.gcodePET_MK3.gcode
Recv: SD printing byte 18722/661767
Recv: 00:06
Recv: ok
Recv: mber is not Last Line Number+1, Last Line: 164
Send: N164 M27*21
Recv: Resend: 165
Recv: ok
Recv: model_0.15mm_PET_MK3.gcode
Recv: SD printing byte 18722/661767
Recv: 00:06
Recv: ok
Send: N165 M27*20
Recv: Error:Line Number is not Last Line Number+1, Last Line: 164
Recv: Resend: 165
Recv: ok
Send: N165 M27*20
Send: N166 M105*38
Recv: model_0.15mm_PET_MK3.gcode
Recv: SD printing byte 19082/661767
Recv: 00:06
Recv: ok
```
</details>

And a few moments later it disconnects with these messages:

<details>
 <summary>Timeout + disconnect</summary>
```
Send: N190 M105*47
Recv: fsensor_stop_and_save_print
Recv: fsensor_oq_meassure_start
Recv: echo:Enqueing to the front: "G1 E-3 F200"
Send: N191 M27*31
Recv: echo:Enqueing to the front: "G1 E3 F200"
Send: N192 M27*28
Recv: fsensor_oq_meassure_stop, 4 samplnd_continue
Recv: 16 yd_sum=134 er_sum=0 er_max=0
Recv:  yd_min=45 yd_max=33 yte 24968/66.4 A:42.2
Recv: model_0.15mm_PET_MK3.gcode
Recv: SD printing byte 24968/661767
Recv: 00:07
Recv: ok
Recv: model_0.15mm_PET_MK3.gcode
Recv: SD printing byte 24968/661767
Recv: 00:07
Recv: ok
Recv: l_0.15mm_PET_MK3.gcode
Recv: SD printing byte 24968/661767
Recv: 00:07
Recv: ok
Recv: model_0.15mm_PET_MK3.gcode
Recv: SD printing byte 24968/661767
Recv: 00:07
Recv: ok
Send: N193 M27*29
Recv: model_0.15mm_PET_MK3.gcode
Recv: SD printing byte 25809/661767
Recv: 00:07
Recv: ok
Send: N194 M27*26
Recv: model_0.15mm_PET_MK3.gcode
Recv: SD printing byte 26157/661767
Recv: 00:07
Recv: ok
Send: N195 M27*27
Recv: model_0.15mm_PET_MK3.gcode
Recv: SD printing byte 26571/661767
Recv: 00:07
Recv: ok
Send: N196 M105*41
Send: N197 M27*25
Recv: ok T:237.3 /240.0 B:89.6 /90.0 T0:237.3 /240.0 @:42 B@:58 P:53.5 A:42.3
Recv: model_0.15mm_PET_MK3.gcode
Recv: SD printing byte 26908/661767
Recv: 00:07
Recv: ok
Send: N198 M27*22
Send: N199 M27*23
Send: N200 M27*20
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N201 M105*36
Send: N202 M27*22
Send: N203 M105*38
Send: N204 M27*16
Recv: echo:busy: processing
Send: N205 M27*17
Send: N206 M27*18
Recv: model_0.15mm_PET_el_0.15mm_PET_MK3.gcode
Recv: SK3.gcode
Recv: SD printing byte 27333/661767
Recv: 00:07
Recv: ok
Recv: model_0.15mm_PET_MK3.gcode
Recv: SD printing byte 27333/661767
Recv: 00:07
Recv: ok
Recv: model_0.15mm_PET_MK3.gcode
Recv: SD printing byte 27419/661767
Recv: 00:07
Recv: ok
Recv: l_0.15mm_PET_MK3.gcode
Recv: SD printing byte 27333/661767
Recv: 00:07
Recv: ok
Recv: model_0.15mm_PET_MK3.gcode
Recv: SD printing byte 27419/661767
Recv: 00:07
Recv: ok
Send: N207 M27*19
Send: N208 M27*28
Send: N209 M105*44
Send: N210 M27*21
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N211 M105*37
Send: N212 M27*23
Send: N213 M27*22
Recv: model_0.15mm_PET_MK3.gcode
Recv: SD printing byte /240.0 @:39 B@:81 P:53.9 A:42.3
See octoprint.log for details
Changing monitoring state from "Printing from SD" to "Offline (Error: See octoprint.log for details)"Connection closed, closing down monitor
```
</details>

jurriaan, 2018-09-24:
Not sure, but could it be that this issue is related to #1187? Since the printer stutters during printing on the latest firmware it could be that this causes the timeouts over USB.

PavelSindler, 2018-09-25:
Thank you for hint. We are testing it now.

PavelSindler, 2018-09-25:
Which Octoprint version do you have? I was testing it with 1.3.9 and it seems reliable so far (both SD card and USB printing). Also what seems strange to me is that in the example above, you are printing from SD card and gcodes which are sent by Octoprint (M27, M105 etc.) are stamped with line number and checksum. It is probably just that I am missing something, but when I print from SD card with Octoprint no line numbers/checksums are present.

jurriaan, 2018-09-25:
Interesting, I'm using the latest version (1.3.9), not sure why those are stamped with line numbers on my end (I only did one test with SD, so I can't say a lot about that).

During your tests did you have these hangs described by https://github.com/prusa3d/Prusa-Firmware/issues/1187? I've had prints where some infill layers were printing with lot's of pauses during the infill. Where you can really see the printer hangs/waits on some input. 

When this happened while printing without Octoprint I recognized that I saw the exact same behavior that I saw during the time Octoprint was timeouting while printing over USB (For the log of those USB print timeouts see https://github.com/prusa3d/Prusa-Firmware/issues/738#issuecomment-422575242)).

I will do some more tests but i think these filament sensor hangs described in #1187 do in some way relate to the issue I'm having since I haven't had this issue since I disabled the filament sensor.

PavelSindler, 2018-09-25:
Can you send me full serial log from octoprint (USB printing)?

PavelSindler, 2018-09-25:
From your logs it seems Octoprint is sending gcode lines without waiting for "ok". I would like to see whole log to see where this behaviour started.

PavelSindler, 2018-09-25:
Yes I had also some "hangs" but it caused no issues. We are testing now also with more problematic filament so I hope there will be some issues. 
However I have tested with prolonged pause where printer checks for filament presence and I was able to trigger communication issues. I have fixed this by sending busy messages properly during this pause (PR https://github.com/prusa3d/Prusa-Firmware/pull/1198), but I am not sure if that is the root cause for your issues.

jurriaan, 2018-09-25:
I could reproduce an RX buffer overflow with Octoprint stopping the print due to corrupted responses from the printer. 

See https://gist.github.com/jurriaan/4e32a5da566f769da4857129b877ae73/raw/4f32d07c0b72a8858b11a4c8dc1f5a0d2527ec03/serial.log

This log file contains a lot of issues, you will see resends, Full RX buffer, weird corruption.

It contains a lot of resends like this (which are corrupted, see the first enqueing message):
```
2018-09-25 19:40:18,521 - Send: N35728 G1 X113.685 Y112.167 E0.01100*99
2018-09-25 19:40:18,577 - Recv: ok
2018-09-25 19:40:18,583 - Send: N35729 G1 X112.090 Y110.571 E0.05878*98
2018-09-25 19:40:18,637 - Recv: fsensor_stop_and_save_print
2018-09-25 19:40:18,640 - Recv: fsensor_oq_meassure_start
2018-09-25 19:40:18,642 - Recv: echo:Enqueing tN35729 Gont: "G1 E-3 F200"
2018-09-25 19:40:19,447 - Recv: echo:Enqueing to the front: "G1 E3 F200"
2018-09-25 19:40:20,347 - Recv: fsensor_oq_meassure_stop, 4 samples
2018-09-25 19:40:20,354 - Recv:  st_sum=716 yd_sum=121 er_sum=0 er_max=0
2018-09-25 19:40:20,407 - Recv:  yd_min=41 yd_max=28 yd_avg=30 sh_avg=17
2018-09-25 19:40:20,411 - Recv: fsensor_err_cnt = 0
2018-09-25 19:40:20,414 - Recv: fsensor_restore_print_and_continue
2018-09-25 19:40:20,417 - Recv: Resend: 35714
```

This is the part that stops the print:

```
2018-09-25 19:48:09,012 - Send: N42380 G1 X148.855 Y63.990 E0.05899*82
2018-09-25 19:48:09,037 - Recv: fsensor_stop_and_save_print
2018-09-25 19:48:09,067 - Recv: fsensor_oq_meassure_start
2018-09-25 19:48:09,069 - Recv: echo:Enqueing to the front: "G1 E-3 F200"
2018-09-25 19:48:09,907 - Recv: echo:Enqueing to the front: "G1 E3 F200"
2018-09-25 19:48:10,807 - Recv: fsensor_oq_meassure_stop, 4 samples_continue
2018-09-25 19:48:10,809 - Recv: Resend:um=116 er_sum=0 er_max=0
2018-09-25 19:48:10,813 - Printer requested line 116 but no sufficient history is available, can't resend
```

You can see that the last response from the printer is corrupted and Octoprint thinks that it needs to send line 116.

Another interesting thing happens at the start of the print:
```
2018-09-25 18:58:53,975 - Send: N27 G21*47
2018-09-25 18:58:53,997 - Recv: Unknown G code: $27 G21
2018-09-25 18:58:54,000 - Recv: ok
```

To me this shows that the filament sensor causes some resends which will probably be fixed a bit by #1198, but there are also weird corruption issues in the serial communication.

If you need more information I'm happy to help.

PavelSindler, 2018-09-26:
Interesting. We did test with 17 hours print with the most problematic filament we found. There were 2331 filament sensor triggered pauses and there was no issue found in log. Can you provide more information about your setup? MK3 printer? Raspberry PI3, Zero or different? Rpi port ON or OFF? Regarding checksum errors we had some issues with very long or low quality USB cables near power supply mains.

jurriaan, 2018-09-26:
I have an MK3, Raspberry PI 3, Rpi port Off, currently using a very short (15cm) USB cable bought from Amazon (https://www.amazon.de/gp/product/B071YJ7LWB/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1) since I positioned the Pi directly above the EINSY board (since I use this case: https://www.thingiverse.com/thing:2334119).

Did you also have resends after filament sensor pauses? Because to me that doesn't directly seem like a USB cable issue (shouldn't the firmware make sure that it doesn't have to ask for resends after the filament sensor checks?). Not that it helps in case the serial communication is corrupted, but I found that interesting as well.

tedder, 2018-10-01:
jurriaan, check to see if you are getting throttling on your Pi (`vcgencmd get_throttled`).

bradcozine, 2018-10-01:
Having issue with recurring printer pauses due to filament sensor printing from Octoprint. Just turned off the filament sensor so I can get this print done. I could post a log, but it's pretty much the same as above. If you'd like one anyway, I'll do one for a smaller print!

- Prusa i3 MK3 3.4.0
- Hatchbox PLA Gold
- Octoprint 1.3.9
- Octopi 0.15.1
- Raspberry Pi 3B+
- Rpi port Off
- Prusa provided USB

JasonRFisher, 2018-10-06:
I started having this issue today on Prusa i3 MK3 3.4.0 and Octoprint 1.3.9.  I updated to 3.4.1 and still experience it. 5-10min into every print it just stops and Octoprint reports timeout and the Terminal tab shows Rx Buffer Overflow.  If I print the same object off SD directly no issue at all.

jurriaan, 2018-11-02:
@tedder currently running Arch Linux aarch64 on my Pi, which doesn't have that command. During printing the load is really low, `load average: 0.03, 0.11, 0.10`, so I don't think that's happening.

@PavelSindler Latest prints did succeed since I turned off filament runout detection, but I still get quite some corruption in the serial port communication which is weird, I find it a bit unlikely that this corruption has anything to do with my Raspberry Pi itself, since USB does use checksums in its communication. Is there a way to debug this?

tedder, 2018-11-02:
There's an "unknown" at the top, but where's the failure at the bottom?

I don't know if Arch has the Pi videocore, but [this snippet may run if you have `/dev/vcio`](https://gist.github.com/tedder/c589de4f092f0b9120c1a41f77ad372a). Returning zero is good.

tedder, 2018-11-02:
Checking the Pi that talks to my Prusa, I see that "unknown g code" occasionally too:

```
2018-11-01 15:31:50,048 - Send: N22 G21*42
2018-11-01 15:31:50,059 - Recv: Unknown G code: $22 G21

2018-11-01 15:32:12,070 - Send: N50 G21*47
2018-11-01 15:32:12,078 - Recv: Unknown G code: $50 G21

2018-11-02 14:46:52,998 - Send: N22 G21*42
2018-11-02 14:46:53,006 - Recv: Unknown G code: $22 G21
```

So it's always on G21. Maybe $22 isn't a sign of corruption? G21 shouldn't be a problem though.

jurriaan, 2018-11-02:
@tedder updated the gist, somehow it was truncated: https://gist.githubusercontent.com/jurriaan/7264a406fa59baf264ca5eb9912400df/raw/f85c67b7a23b506eae43d0bac4f805f0904f0bd7/serial.log 

At the top you can also see some weird characters being received by the pi

jurriaan, 2018-11-02:
@tedder Unfortunately no /dev/vcio here.

jurriaan, 2018-12-12:
@PavelSindler anything else I can try to get this to work?

Arksine, 2018-12-24:
Hello, I thought I would chime in here and hopefully help you out.  I maintain the PrusaOwners soft fork of [Klipper,](https://github.com/KevinOConnor/klipper) which absolutely requires stable serial comms. After many users, including myself, experienced random disconnects I decided to research the issue and found two contributing factors that lead to disconnections.  After fixing them I have zero data corruption and a rock solid connection at 250000 baud.

First, and the most glaring, is the arduino-usbserial firmware flashed on the Atmega 32u2.  Assuming the Einsy uses [this repo](https://github.com/ultimachine/rambo-32u2-usbserial), its based on old code, and there are known issues with serial data corruption.  See this [pull request](https://github.com/arduino/Arduino/pull/4252) for an example.  To fix this I flashed the Atmega 32u2 with [Hoodloader2](https://github.com/NicoHood/HoodLoader2).  I recognize that this is beyond the comfort level of most users, and Prusa isn't likely to recall every Einsy to flash improved usbserial firmware, however going forward I would strongly recommend taking a look at the usbserial firmware and updating it to something more robust.  This may be something Prusa needs to work with Ultimachine on.

In the meantime there is another contributing factor that when fixed should result in stable communications at 115200 baud without flashing the 32u2.  Prusa's custom stk500v2 bootloader enables UART2, however it doesn't disable it before loading the main program.  I don't know exactly why (noise causing a buffer overflow?), but after time it seems that this can lead to disconnections on UART0.  Simply disabling UART2 in Prusa Firmware's setup() routine should resolve this issue.  It is probably wise to disable all UARTs in setup() and only re-enable when used.

Arksine, 2018-12-25:
@jurriaan I installed using a secondary [Arduino Uno with installation_sketch.ino](https://github.com/NicoHood/HoodLoader2/wiki/Burn-the-bootloader-from-another-Arduino), however it shouldn't be difficult using a programmer and the Arduino IDE.

1) Follow [these directions](https://github.com/NicoHood/HoodLoader2/wiki/Software-Installation) to install the boards.
2) Make sure printer is unplugged, both power and usb. Also make sure the programmer is NOT connected via USB yet.
3) Hook up your programmer to the Einsy's 32u2 ICSP header.
[image]
In this picture the reset pin (pin 5) is the the bottom rightmost pin.
3) Plug in your programmer
4) Select Hoodloader2 32u2 from the boards manager
5) Select Hoodloader2 Mega in the Board / Bootloader /USB-Core
6) Select your programmer
7) Click Burn Bootloader

I strongly recommend reading the entire Hoodloader2 wiki before proceeding.
Alternatively you can use an Arduino Uno as outlined in Hoodloader2's [Installation sketch instructions](https://github.com/NicoHood/HoodLoader2/wiki/Burn-the-bootloader-from-another-Arduino).

Some things to keep in mind:
- I haven't tested this with Prusa Firmware, only Klipper.  It does behave identically to the previous firmware when it comes to flashing the Mega, so there should be no issues.
- One of the features of Hoodloader is the ability to load sketches to the 32u2.  We aren't really interested in this, however Hoodloader uses 57600 baud to load those sketches, so we cant use that baud rate to connect to anything (much like we cant use 1200 baud, because that is the baud rate used to flash firmware to the Mega 2560)
- This will change your product and vendor IDs.
- After you do this you will overwrite the original usbserial firmware on the 32u2.  If you want to revert you will need to either dump it with the programmer beforehand or request it from Prusa.

jurriaan, 2018-12-25:
@Arksine thanks for the explanation! Which USB serial firmware do you use? Or does the Hoodloader2 firmware handle this by default? I will try this in the coming days.

And what about the custom stk500v2 bootloader enabling the extra UART. Can I fix that as well?

Arksine, 2018-12-25:
@jurriaan  Hoodloader2 has its own built-in fast usbserial implementation, so it handles it.  

The easiest way to fix the UART2 problem is to disable in the the setup() function in Prusa Firmware.  If you are set up to build Prusa firmware, just put the following at the top of the setup() function in marlin_main.cpp:

```UCSR2B = 0;``` 

That will disable UART2.

Arksine, 2018-12-26:
@Panayiotis-git it can be re-enabled when the MMU is connected.  It shouldn't be enabled when not in use.

Panayiotis-git, 2018-12-26:
@Arksine You are right. I tried the "UCSR2B = 0;" after the mmu_init() and the result was a not responding MMU unit.

jurriaan, 2018-12-27:
Updated the firmware of the 32u2 to Hoodloader2 so far it seems to be working, trying some prints with Octoprint right now. Tnx @Arksine! 

Didn't disable the UART2 yet, since I didn't feel like recompiling the Prusa firmware ;). I do however think UART2 could be enabled in the official version if it isn't used for printers without an MMU (cc @PavelSindler).

Will do some test prints in the coming days and let you know how this pans out :)

jurriaan, 2018-12-27:
I did finish a three hour print using Octoprint and Hoodloader2 without any error messages! 

One thing I did however notice is that when using Octoprint is that it runs pretty slow using more complex infill patterns like gyroid (I think since it generates quite complex gcode) and I also noticed some jagged lines along rounded corners of the skirt (which seems a bit weird to me). Not sure why this happens, but these things result in the print being slow and not that pretty. 

I will try some more Octoprint prints in the future, maybe even with tweaks to baudrate and serial buffer sizes, but for now I'll just fall back to SD printing, since printing over USB just doesn't work as seamless as printing from SD.

pleedom, 2019-01-03:
I am currently unable to connect Octoprint to the Prusa MK2.5 (Firmware 3.4.1) unless the Filament Sensor is set to "ON" in the setttings. Idk if this is helpful for this issue or not, but it might be related.

jurriaan, 2019-04-01:
Update on this, currently been running with Hoodloader firmware for quite some time, only doing SD card prints, but no errors or hangs anymore, where before it errored even during SD card printing after printing for a few hours. 

To me it really feels like the serial bootloader needs an update here as @Arksine described.

cc @PavelSindler / @mkbel

deejay2, 2019-04-01:
I posted it in a another issue that was closed but lead to open this one. Tell me if I'm off topic...  Here's my log:

I have the same issue on 3.6.0 :

2019-03-30 18:25:14,107 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-03-30 18:25:44,186 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-03-30 18:26:14,239 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-03-30 18:26:44,286 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-03-30 18:27:14,365 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-03-30 18:27:43,213 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-03-30 18:27:44,407 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-03-30 18:28:14,486 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-03-30 18:28:41,265 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2019-03-30 18:28:41,308 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 152051, current line = 152057
| Last lines in terminal:
| Send: N152048 G1 X147.716 Y75.254 E0.17206105
| Recv: ok
| Send: N152049 M204 S1000108
| Recv: ok
| Recv: ok
| Send: N152050 M117 99% L=52/52108
| Send: N152051 G1 E-0.08000 F1800.000005
| Recv: ok
| Recv: ok
| Send: N152052 G1 F8640117
| Recv: ok
| Send: N152053 G1 X150.159 Y72.810 E-0.6840064
| Recv: ok
| Send: N152054 G1 E-0.03600 F1800.0000013
| Send: N152055 G1 Z10.800 F10800.00050
| Recv: Full RX Buffer
| Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
| Send: N152056 M105*18
| Recv: Error:Line Number is not Last Line Number+1, Last Line: 152050
| Recv: Resend: 152051
2019-03-30 18:28:44,524 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-03-30 18:29:14,596 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-03-30 18:29:16,149 - octoprint.util.comm - INFO - Finished in 13282.713 s.
2019-03-30 18:29:16,151 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Finishing"
2019-03-30 18:29:16,193 - octoprint.plugins.DisplayLayerProgress - INFO - Printing stopped. Detailed progress stopped.
2019-03-30 18:29:16,291 - octoprint.plugin - ERROR - Error while calling plugin tracking
Traceback (most recent call last):
File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/init.py", line 230, in call_plugin
result = getattr(plugin, method)(*args, **kwargs)
File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking/init.py", line 119, in on_event
self._track_printjob_event(event, payload)
File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking/init.py", line 237, in _track_printjob_event
sha.update(self._settings.get([b"unique_id"]))
TypeError: update() argument 1 must be string or buffer, not None

Arksine, 2019-04-02:
For what its worth, I did Prusa's work for them.  See [this repo](https://github.com/PrusaOwners/mk3-32u2-firmware).

The mk3-32u2-firmware contains a port of Hoodloader's fast usbserial and combines it with a recent version LUFA's DFU firmware.  I haven't actually tested the DFU portion as it requires removing the Einsy from the case and jumpering some pins, but it is unchanged from LUFA's standard implementation so it should work. 

The above firmware keeps Prusa's vendor and product strings, as well as the custom serial ID.  At the time I flashed hoodloader I didn't realize that the Serial Number was stored in EEPROM, however I had recorded it myself.  @matthew-humphrey was kind enough to dump his EEPROM and reverse the location and encoding for me.

jurriaan, 2019-04-02:
@Arksine Awesome, will try to flash that soon!

Oh :sweat_smile: hopefully I didn't wipe my EEPROM then. I hope someone at Prusa can looks into your repository. I'll also send support a message about this (they won't really help with Octoprint related stuff, but I can try to at least tell them the usb-serial implementation is faulty.)

matthew-humphrey, 2019-04-02:
Nice work, @Arksine. Will flash this on my printers this evening.

fawick, 2019-05-09:
For the users who have installed https://github.com/PrusaOwners/mk3-32u2-firmware: Did you observe any disconnects during prints afterwards?

@josefprusa Is there any chance this fix is going to be released officially from Prusa3D?

kanzelsberger, 2019-05-14:
I'm having frequent USB disconnects with Octoprint and temporarily switched to Klipper. Now I found this so I might give this a try. I hope it won't interfere with newly released fw 3.7.1

geeksville, 2019-12-25:
I just wanted to add a datapoint that I see this problem often with Octoprint talking to my MK3S+MMU2S.  My firmware is 3.8.1 (and 1.0.6 on the MMU).  Is it possible that all the extra comms via UART2 (to the MMU) makes the problem more apparent in this config?

The failure only seems to occur if I'm doing filament changes, if just using the printer in single color mode it doesn't happen.

@PavelSindler & @Arksine have others confirmed that the new firmware for the avr USB-serial bridge fixes this?  I can build and program mine but my printer is buried in a lack enclosure (with separate powersupply) so I sort of dread getting access to that header for programming.

typical octoprint log when the problem occurs:
```
2019-12-25 04:18:27,730 - octoprint.util.comm - INFO - Ignoring resend request for line 355063, that still originates from lines we sent before we got the first resend request
2019-12-25 04:22:52,717 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 357010, current line = 357015
| Last lines in terminal:
| Send: N357006 G1 X122.636 Y110.451 E-0.10125*125
| Recv: ok
| Recv: ok
| Send: N357007 G1 X122.964 Y110.503 E-0.07657*118
| Send: N357008 G1 X123.073 Y110.544 E-0.02695*127
| Recv: ok
| Send: N357009 G1 X123.292 Y110.678 E-0.05909*114
| Recv: ok
| Send: N357010 G1 X123.524 Y110.910 E-0.07590*127
| Recv: ok
| Send: N357011 G1 X124.624 Y109.162 E-0.47666*113
| Recv: ok
| Send: N357012 G1 X124.430 Y109.180 E-0.04494*113
| Recv: ok
| Recv: Full RX Buffer
| Send: N357013 G1 E-0.07000 F2100.00000*1
| Recv: ok
| Send: N357014 G1 X122.080 Y111.826 F10800.000*88
| Recv: Error:Line Number is not Last Line Number+1, Last Line: 357009
| Recv: Resend: 357010
2019-12-25 04:22:52,736 - octoprint.util.comm - INFO - Ignoring resend request for line 357010, that still originates from lines we sent before we got the first resend request
2019-12-25 04:22:52,747 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 357010, current line = 357015
| Last lines in terminal:
| Send: N357011 G1 X124.624 Y109.162 E-0.47666*113
| Recv: ok
| Send: N357012 G1 X124.430 Y109.180 E-0.04494*113
| Recv: ok
| Recv: Full RX Buffer
| Send: N357013 G1 E-0.07000 F2100.00000*1
| Recv: ok
| Send: N357014 G1 X122.080 Y111.826 F10800.000*88
| Recv: Error:Line Number is not Last Line Number+1, Last Line: 357009
| Recv: Resend: 357010
| Recv: ok
| Send: N357010 G1 X123.524 Y110.910 E-0.07590*127
| Recv: ok
| Send: N357011 G1 X124.624 Y109.162 E-0.47666*113
| Recv: Error:Line Number is not Last Line Number+1, Last Line: 357009
| Recv: Resend: 357010
| Recv: ok
| Recv: Error:No Line Number with checksum, Last Line: 357009
| Send: N357012 G1 X124.430 Y109.180 E-0.04494*113
| Recv: Resend: 357010
```

gbeatty, 2022-02-09:
Does anyone know if this is still an issue in the latest 3.10.1 firmware?

closed 2023-09-11 by github-actions[bot]