prusa3d/Prusa-Firmware #1181: MMU2 Gets Confused About Selector Bearing Position

pleppik, 2018-09-20:
Occasionally my MMU2 selector bearing will get confused about its position.

When loading or unloading a filament, the bearing moves to a position which does not engage the correct filament. Sometimes the bearing will engage the wrong filament, causing it to be ejected out the front of the MMU. Other times the bearing will only partially engage or not engage on any filament, causing a load or unload failure.

This happens maybe once every 1000-2000 material changes. When it happens, resetting both the MMU and the MK3 using the reset buttons next to the USB ports will correct the problem. Cycling the power or resetting the MK3 with the front panel button does not reliably fix the issue.

The selector bearing can travel freely to both stops.

ojrik, 2018-09-20:
The screws with the string should be aligned or as shown in the picture without a filament.
[image]

pleppik, 2018-09-20:
The MMU is properly tensioned. The problem is that the selector bearing is not being positioned properly.

For example, I select "Load Filament 2", and the selector moves to the position to accept filament 2, but the idler body moves to engage a different filament (for example, 4). The result is that filament 4 is ejected out the front of the MMU.

This happens only rarely, but when it happens the only reliable way I've found to clear the problem is to reset the MMU and MK3 using the reset buttons near the USB ports. If I simply cycle the power and allow it to rehome, the MMU will continue mispositioning the idler body.

Blanchard3d, 2018-09-23:
Like this? Mine does all kinds of crazy stuff, its driving me nuts. I'm starting to think the drivers or board might be bad. 

[image]

chrissearle, 2018-09-23:
I've had this twice today (first day with the MMU). First time I gave up and powered down.

Second time - I got support involved.

It happened the second time after a filament load/unload issue. This turned out to be a stringing filament under the finda.

The print was a two colour print using 1 and 5. After I cleared the finda - it suddenly gave me a filament load flashing LED on position 5 when trying to load filament 5 - with the selector on position 5 - but with the idler body positioned for filament 1.

We tried a fair number of things - but it insisted on pushing the filament in position one out just like @Blanchard3d  pic in comment above.

So - we decided to try one last thing - while the printer was still showing "MMU needs user attention" - I hit the reset of the MMU only.

It reset, selector moved to position 1,  and the MK3 suddenly changed back to heating nozzle.

Once the heating was complete - it then pushed the filament out of position 5 with the selector in position 1 and started to beep. But this time the LCD asked me to remove the old filament and then press the knob on the mk3. 

I did this - and the unit then did the proper reload of filament 5 (moved the selector to the right place) and the print continued. Still printing now - just hoping it'll complete.

So - I have also seen this, it's definitely that the MMU idler and selector are in disagreement on position - one is one place - the other a different place.

Seems like a definite bug in the MMU firmware.

But - it does seem - as a last resort - that resetting the MMU alone while the printer holds the progress is worth at try - am guessing only as a last resort as I have no idea how reliable this is - but if the print is going to fail anyway - worth a try :)

AbeFM, 2018-09-30:
This happens to me.

It could be solved, I feel, by just homing the drum more often. Maybe every layer?

Similarly, it can lose track of the output selector, and print with the wrong filament. I have this once in a while, though less since I cleaned up the filament paths. Nonetheless, the wrong sized bondtech gears with a single calibration eventually fill the printer with plastic scrap, and causes any number of jams.

AbeFM, 2018-10-18:
The drum commonly gets lost if it skips, and it skips too easily. Monitoring it is likely a better solution than turning up the current.

infomagichand, 2018-11-01:
Im gettins the same problem!
every 1 to 3º filament change it get lost

dikodahan, 2018-11-01:
Saw the same thing. Usually 1-3 moves without an issue, but over to 4 and 5 it sometimes has an issue.
I have to say that if I manually press to move one position at a time, I did not see this issue. However, If I hold the move position to move all the way from 1 to 4 (for example) it gets confused.

jlmyra, 2018-12-28:
I have experienced this (pulley barrel getting out of sync with the selector) as well twice in the last 24 hours. I was unable to clear it either time. I have the latest firmware installed- 3.5.1 and 1.0.3. Very frustrating to lose prints after a huge time and filament investment.

YoungR0se, 2019-01-01:
My first multi-colour print (2 colours per layer) was 9 layers in when the drum misalligned.  Has been unable to align properly for filament change since then.  Tried power cycling, resetting, re-flashing. Nothing makes it work any better :(

martysweet, 2019-01-24:
I had this at the end of a print, and during the second filament change of the next print (after a power cycle). The idler looks to be off by -1 place. 

Removing the filament manually, then pressing the reset button on the MMU2 for 10-15 seconds caused the MMU2 to reboot and re-calibrate, and allowed me to continue the print without restarting the printer itself.

AbeFM, 2019-01-24:
That's pretty interesting. I've had times where I cannot get the drum to
home, and most of those times, I find it goes away when I get all 5
filaments fully loaded or fully unloaded. When some are loose above the
gears, I feel they get pushed to the side, and then they mess with the
homing by adding resistance during that procedure.
Your experience seems contrary.

eschlenz, 2019-01-25:
This issue started happening to me as well, but only with one specific model. I've tried to print the model 3 times now, and each time it appears to fail at the same spot which is roughly 40% of the way to completion.

The result of the failure is the same as everyone described. The idler is off by one compared to the selector. I can reset the MMU2 to get things going again. But then it starts printing without actually having any filament loaded.

I've been working under the assumption that it was either a Slic3r bug with the generated .gcode. Or, a bug with the MMU2/printer in translating the .gcode to instructions for the MMU2. I came to that conclusion because of the fact that it seems to consistently exhibit this behavior at the same point of the model I'm trying to print.

It's probably also worth noting that, in my case, the issue appears after many color changes across all 5 selector positions.

eschlenz, 2019-01-25:
@martysweet I'm not sure. But I can try doing the print again and either try to catch it when it happens. OR, I believe the lights blink at different speeds depending on if it encounters a load/unload issue, correct?

martysweet, 2019-01-25:
@eschlenz Yes that's correct. When it's happened to me the filament will still be in the hotend and the correct LED will be fasting flashing red on the MMU2. Slow flashing for load. Fast flashing for unload. At that point, any button presses engages the wrong selector. Page 29 of the Mk3 MMU2 handbook for troubleshooting, which is where I noticed this issue.

AbeFM, 2019-01-25:
It would be helpful for those not logging to have the current z height (or
g-code line number?) displayed on the screen at all times to make it easier
to figure out what commands the printer was getting.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

LeileDev, 2019-02-06:
I had the same issue until today. I watched what was happening, and realized that the idler was squeezed in too tight and was unable to rotate.  I loosened the two screws shown in the photo below, and everything _seems_ to be working at this point.

[image]

Good luck!

AbeFM, 2019-02-06:
> I have found that the screw on the right in your photo can be and maybe
> even should be tighter than the screw on the left. If the screw on the left
> is too tight the idler tumbler will not rotate to position five, the
> position furtherest from the motor.
>
> Unfortunately, it requires way too much experimentation to zero in on the
> correct tension of these screws. Then if you ever need to open the MMU2 you
> get to do it all over again. If the screws are too loose they won’t load or
> unload the filament. The suite spot is difficult to find and the answer is
> not setting the screw heads flush with the surface or just below the
> surface. I wish the design could be improved but the don’t have a better
> solution.

It can be very loose and still work. I've also had the motor shaft not
parallel to drum and it'll be tight in spots from that.

Can't you just loosen it? I run both screws flush with top most of the
time. I may try tightening right-hand one.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

GitPetri, 2019-03-14:
I see this problem constantly, just every now and then the drum position get unsynchronized compared to the filament feeder (part containing PINDA) and/or selector unsynchronized against MMU main unit expectation, FW 1.0.3. Pressing X-reset on the LCD controller normally doesn't fix this but I've been able to align the parts by using the right filament selector button on MMU2. Of course this makes multi material printing virtually impossible as 90% of my MM print attempts fail due this. MMU2, imho, is not a reliable at the moment and sad to say I totally regret purchasing it.

ainuke, 2019-05-10:
I am currently having this problem after upgrading to mk3s/mmu2s and updating the firmware. To be clear, though, I had the mmu2s upgrade in place and working reliably, and just upgraded the extruder end to mk3s and updated firmware to 3.7.0 for mk3s. 

In addition to misalignment of drum/selector, when the two *are* aligned (such as after using reset button on mmu to fix the misalignment), the drum will rotate a few degrees back and forth at the same filament position, almost like dipping your toes in the water at a pool to check the temperature. 

I’ve checked for unobstructed path; I can open the drum carrier and manually feed the filament into the Bowden tube a few inches, easily. However, if I close the drum carrier and try “load filament to nozzle” on *any* filament position, it fails to load into the tube at all. 

This issue started 20% into a 13 hour three color print from SD. The first couple load failures were correctable just by acknowledging the failure message on the LCD. There were 2 or 3 recoverable errors before the “cant figure out how to load the filament” problem described above settled in. Ultimately, the print had to be aborted. 
This seemed to be in that ~100 filament changes range described in an earlier post; *and* the drum motor seems to be hotter than it should be. (Not sure exactly how hot, but similar or a little warmer than the print bed, so 65-70C?)
Maybe there’s a thermal problem with the motor causing failures?

jsgphd, 2019-05-12:
I'm having the same issue...  Between the idler miss-positioning and the filament frequently getting stuck in the main tube this unit on unloads, this unit has been nothing but problems.  I am used to kicking off a print and just expecting it to finish.  The MMU needs constant babysitting and maintenance every few layers.  I am very disappointed with the unit.  Its a prototype that is not even close to being ready for production, but sold with no warning to that effect...  I am not using it for multi-color printing anymore because I can't make baby sitting a printer my full time job, so I paid $300 to save me the 1min of time changing filiment between single color prints...   pitiful, PRUSA should offer a refund to everyone or quickly issue a new version that doesn't have this many issues.  From the amount of people complaining about this product, I seriously doubt more then a tiny percentage of users are getting regular and successful multi-color prints on a reliable basis.

TheNore, 2019-05-16:
I am also experiencing this issue, the selector tumbler will be misaligned at random times, on prints that have multiple colors (from 2 to 5), all layer heights, and all types of filament, including prusament. I am on firmware 3.7.1 and 1.0.5 for mmu. all of my MMU prints fail bc of this issue, the only way I can resume the halted print is to reset the mmu via button next to the usb. For my experience, this issue exactly mirrors all issues described above.

rambopierce, 2019-05-16:
You might try loosening the right (looking from the front of the printer) screw on the MMU2. I found that there is a hard to find sweet spot on the MMU2 screw tension.  Too loose, the filament doesn’t feed properly, too tight and the tumbler doesn’t always turn properly. The one on the left closest to the tumbler motor tolerates being tighter than the one on the right. I wish the design could be updated so that there is a more exact tension setting.

TheNore, 2019-05-16:
I will try this, as of now I reset the MMU2 and loosened the springs, and it has improved the Loading/ unloading tremendously.

realjordanna, 2019-05-30:
I'm seeing this too out of the box, have not even tried loading a model yet.

AbeFM, 2019-05-30:
https://www.thingiverse.com/thing:3282939 I've had this help if the drum is actually loosening, which can be accelerated by high temps and constant rehoming.

I'm unclear why Prusa rejects input from the community on how to improve their products.

jendit, 2019-06-01:
> I've had this twice today (first day with the MMU). First time I gave up and powered down.
>
> Second time - I got support involved.
>
> It happened the second time after a filament load/unload issue. [...]

Had the same issue today. The MMU should load F3 but the bearing on F2 was trying to push it down/through.

Tried a lot of forth and back, opening, cutting ends, moving manually ...
The only thing that helped, in the end, has been to reset the MMU alone while the print was still active (was at 88% of a 17-hour print).
Although I had to select the right filament manually afterward as the printer did not select the right one on its own. After that, it worked fine again. Also, it seemed to me the unload / load to nozzle was a lot faster afterward again ... maybe some memory issue?

vbtiger, 2019-06-09:
I have the exact same issue. I got the MK3 to MK3S MMU2S upgrade last week. I assembled it and have this issue at like every 10th filament change. It is unacceptable.
Either it is a firmware problem or someone please tell me what can cause this unsynchronized selector? It's driving me crazy. I've tried to print a really big 3 color model, like 20+ hours, but I have to manually fix it like every 30 minutes.

GitPetri, 2019-06-10:
I have this problem occurring constantly. I personally suspect MMU2S firmware problem.

jendit, 2019-06-10:
Well, the speed increase must have been the switched off MMU silence mode even though it still was showing silence mode. Switching to normal and back to silence and the speed has been the same as before for me.

Further on, I had some successful prints since then with only a few MMU load/unload fails. Even one 16 hours print with 0 fails at all :)

What was causing a lot of issues to me was the not perfect IR calibration (fixed that) and too tight feather screws at the MMU and the extruder.

The failure with the position of the selector at the wrong filament did not occur for me since then again ... but expecting it to show again at some point as nothing has been changed regarding this issue.

SaltyPaws, 2019-10-19:
I have the same issue, where the selector is engaged in the wrong filament. Load filament works for filament 1-5. However, loading to nozzle only works for 1 and occasionally 2. For filament 3-5 the selector is consistently in the wrong position. I am on firmware FIRMWARE 3.8.0 (1.0.6)
Things I have tried to remedy:
Change spring tension
Reset mmu2
Reset mmu2 and prusa mk3s
Power cycle
Reflash firmware

So far nothing has fixed the issue. I have had successful prints using the mmu before using filament 1-3 for the same print, this issue appeared one day.

realjordanna, 2019-10-19:
It's been months for me and no luck, gonna just undo the upgrade at this point.

TheNore, 2019-10-20:
you also need to make sure that the idler selector is positioned correctly
on the gears. the idler gears had to be adjust for me 4 times. Each after I
changed the spring and idler positions. this is a large overlook for some
mmu2 owners, and easly the most important adjustment to get perfect

SaltyPaws, 2019-10-20:
@TheNore, this could be the silver bullet. Could you please be specific with what it means to position the idler selector correctly on the gears? What does the right way look like, and what does the wrong way look like?

TheNore, 2019-10-21:
This is first:
https://manual.prusa3d.com/Guide/5.+Pulley+body+assembly/1040?lang=en#s18168

then to tune a little further...
Loosen these screws -
https://manual.prusa3d.com/Guide/4.+Idler+body+assembly/1039?lang=en#s18147

and make sure the entire idler is spaced a few mm away from the motor here:
https://manual.prusa3d.com/Guide/4.+Idler+body+assembly/1039?lang=en#s18155
Pay attention to this part, this is what solved the issue for me.
"Ensure the bearings are in the middle of the grooves in the idler-body and
you can rotate the idler easily, if not slide the entire idler."

SaltyPaws, 2019-11-03:
It turned out to be a hardware in my case. The selector shaft was slipping. Positioning the shaft, and tightening the 2 screws resolved the issue

AbeFM, 2019-11-04:
I really trusted mine, since I was using a metal gear inside the drum to
prevent slipping, but to my horror, I found even the metal gear's set screw
backed out.

I tightened mine back up and things are happy again.

Probably this deserves a new thread - but could a check be implemented such
that if a couple loads go back in a row, you run the drum full CW, full CCW
a couple times, count the steps in between, and check for a loose screw by
detecting motor/drum slop?

paullmullen, 2019-11-19:
Ditto.  Loose set screws.  Easy fix.

markus-i, 2019-11-24:
I'll be d*mned. Loose set screws, also here. And of course the selector drum programming is designed to shake those screws loose by repeatedly ramming the drum against the stops in order to determine the zero position.

kimpixel, 2020-05-17:
Hallo
I can sadly confirm this behavior.
After long prints the system isnt working. 

The Idler-Drum-Motor skips steps and after this the mmu is not able so grap a single fillament anymore.

i dont know to reset/reinit the mmu during the print.
Until yesterday i break all my Prints.

**Now i notice the drum can be rotate by hand (with a little force) and korrect the skipped stepps.
But i neet some trys to find trhe right possition.**

If i had a way to re-calibrate the sytem durring the error/pause it will much easier

paullmullen, 2020-05-17:
I have seen this problem.  Usually it’s a set screw (left side of the idler drum) that is loose.  I have tightened it once and never again had a problem.  Hope that helps.

 

Paul

kimpixel, 2020-05-17:
@paullmullen
I had this error too - but in this case this was not the problem

TheNore, 2020-05-18:
I have also solved this by unplugging the power to the mmu board for 39
seconds. This error is sometimes caused by the motor being underpowered.

mmierke, 2021-06-02:
Looks to me like this happens when the filament passed the Finda, but prevents the selector to move.

GitPetri, 2021-06-03:
> Looks to me like this happens when the filament passed the Finda, but prevents the selector to move.

I don't believe that is the only problem as this incident every now and then happens in mid-print without any filament changes. Filament offloading just fails.

mmierke, 2021-06-07:
> > Looks to me like this happens when the filament passed the Finda, but prevents the selector to move.
> 
> I don't believe that is the only problem as this incident every now and then happens in mid-print without any filament changes. Filament offloading just fails.

This is at least one symptom I could identify. Would be cool to have a possible way to manually adjust idler and pulley. I don't like the fact, that the GUI is blocked whenever the mmu needs attention.

closed 2023-09-11 by github-actions[bot]