prusa3d/Prusa-Firmware #1184: octoprint "Printer reset detected" warning

rdavolio, 2018-09-21:
I recently flashed the latest firmware (3.4.0)
I am getting a warning "Printer reset detected" when I open Octoprint.
The warning says:"It looks like your printer reset while a connection was active. If this was intentional you may safely ignore this message. Otherwise you should investigate why your printer reset itself, since this will interrupt prints and also file transfers to your printer's SD."

I posted this error on the Octoprint forum and I have been told its a firmware issue.
Here attached are the log files.
[octoprint.log](https://github.com/prusa3d/Prusa-Firmware/files/2405374/octoprint.log)
[serial.log](https://github.com/prusa3d/Prusa-Firmware/files/2405375/serial.log)
[TerminalTab.txt](https://github.com/prusa3d/Prusa-Firmware/files/2405376/TerminalTab.txt)

foosel, 2018-09-21:
To add on to this, for some reason the firmware appears to be sending `start` twice - which it shouldn't. See this excerpt from the serial communication log:

```
2018-09-21 11:43:03,294 - Changing monitoring state from "Offline" to "Detecting serial port"
2018-09-21 11:43:03,357 - Serial port list: ['/dev/ttyACM0']
2018-09-21 11:43:03,357 - Connecting to: /dev/ttyACM0
2018-09-21 11:43:03,366 - Changing monitoring state from "Detecting serial port" to "Opening serial port"
2018-09-21 11:43:03,370 - Connected to: Serial<id=0x6daafc10, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
2018-09-21 11:43:03,371 - Starting baud rate detection...
2018-09-21 11:43:03,371 - Changing monitoring state from "Opening serial port" to "Detecting baudrate"
2018-09-21 11:43:04,380 - Trying baudrate: 115200
2018-09-21 11:43:04,404 - Send: N0 M110 N0*125
2018-09-21 11:43:04,713 - Recv: start
2018-09-21 11:43:04,717 - Changing monitoring state from "Detecting baudrate" to "Operational"
2018-09-21 11:43:04,729 - Recv: start
2018-09-21 11:43:04,732 - Printer sent 'start' while already operational. External reset? Resetting line numbers to be on the safe side
2018-09-21 11:43:04,733 - Send: N1 M105*38
```

The `start` sent at 2018-09-21 11:43:04,729 is extra and makes it look to OctoPrint (or any host sensitive to `start` messages after an established connection for that matter) as if the printer just reset.

Or maybe it is actually resetting right after startup, in which case it should be investigated why that is the case.

dikodahan, 2018-09-21:
I have noticed that if the printer is on, and it even happened to me once in the middle of a print job, and OctoPrint is rebooting it will reset the printer.

closed 2023-09-11 by github-actions[bot]