prusa3d/Prusa-Firmware #1186: Possible cause for the print pauses people are seeing in 3.4.0

geeksville, 2018-09-22:
Dear Prusa swengs,

I just changed my printer to use a bearing to indirectly read filament movement (thus solving the problems with translucent filament): https://www.thingiverse.com/thing:3091625

But in the process of tweaking the code for this change I accidentally stumbled upon the behavior that some people are reporting on your forum.  I had never seen this problem occur on my printer before, so while it was happening I think I discovered at least the area of the code that has the bug.  When the pauses occurred I happened to have a debug console open on the serial port and noticed this gcode getting inserted:

https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/fsensor.cpp#L454

So I poked around a bit, and sure enough - every time it paused it was because this fsensor_watch_runout check had fired (when it shouldn't have - because the filament was feeding fine and the ISR should have been not increasing fsensor_err_cnt).

Therefore I think you might want to investigate one of two possible things:

1) https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/fsensor.cpp#L13
Is FSENSOR_ERR_MAX too low and therefore for some users/printers you might be invoking the meat of fsensor_update() too often? - because if that code is run the print will pause.

2) What is the intent of https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/fsensor.cpp#L347 ?  It seems like you are trying to say "if things have started to fail, make the detection happen faster."  But I don't understand why you'd want to magically increment by 2 vs (some other number or the 1 you normally use)?  It seems to me that makes the incrementing of fsensor_err_cnt asymmetric with the code you use to back off from the faults.  So if you are unlucky you might reach FSENSOR_ERR_MAX when that was not the intent.

But I didn't really look at the code carefully (because once I fixed my issue with the bearing I didn't need to ;-)) and I don't know a lot about Marlin.  However, I did want to send you this note in case it helps.

PavelSindler, 2018-09-24:
Thank you @geeksville . Yes it seems that pauses are caused by too sensitive filament setection (probably related to https://github.com/prusa3d/Prusa-Firmware/issues/1179). We are now doing some tests to compare filament detection sensitivity for 3.3.1, 3.4.0 and few modified 3.4.0 versions.

geeksville, 2018-09-28:
@PavelSindler sounds good.  Any thoughts on theory #2 above?  It seems suspicious for me and I not sure of what the author intended.  Possibly have an internal code review inside of Prusa?

3d-gussner, 2020-09-22:
@geeksville Is this still an issue with newest firmware? If not please close this issue.

JohnnyDeer, 2020-09-29:
I´m closing this issue as stale.

closed 2020-09-29 by JohnnyDeer