prusa3d/Prusa-Firmware #1187: [3.4.0] Filament Engine 2.0 Print Stutters

sarf2k4, 2018-09-22:
I noticed on several prints by about 2-3 prints, my printer periodically paused by 1 second then continues.

**Behavior**
By having filament sensor turned on and printer is printing any solid infill be it bottom, top (I am sure this is also affected), angled perimeter solid (from slic3r's "ensure wall thickness") and maybe similar, the printer prints like normal but then paused for 1 second then continue like nothing happened. 

**Solution**
By disabling filament sensor from setting/tune menu, the pause issue were gone. This has been confirmed by monitoring several layers that has partial solid infill between filament sensor turned on and turned off. 

**Theory**
With improved filament engine 2.0, the firmware periodically checks the filament sensor's visibility and whether or not the filament is still moving while printing. I believe that the filament engine 2.0 has its own buffers that can't keep up with the filament velocity such as solid inflill where the filament goes in continuously except perimeters.

I'm not sure about the theory and all.

Please fix this issue

ThaliaFromPrussia, 2018-09-24:
This is what happened the last prints to me as well. I hope this gets fixed.

TheAdamizer, 2018-09-24:
Hmm I get these pauses on 3.4.0, even though I leave filament sensor off all the time (not reliable enough to warrant the risk it poses with a false negative).  I just have to be printing over usb serial from octoprint.  These would end up being permanent print crashes until I bumped the octoprint timeout tolerance from 6 to 10 times.  Now they are just 2 second or so pauses.

PavelSindler, 2018-09-25:
Pauses during print when filament sensor is ON are ok. If sensor detects possible filament runout, printer pauses print and tries to move filament backwards (something like retract) and back while measuring response. Then it decides if filament is still present or not. If filament is still there, printer continue printing. If not, M600 (filament change) is run.

PavelSindler, 2018-09-25:
You can check if E-stepper / bondtech is moving during pause. If it is moving, it is ok and this behaviour is desired.

sarf2k4, 2018-09-25:
Actually this is abnormal for us 3d printer owners because pausing for 1 second for every solid infill, it would make the print time longer and thus create inaccurate printing time. Not just that, it may affect print quality especially the overhang where the heat will creep towards the printed parts overhang and create curling there.

3d-gussner, 2020-09-22:
@sarf2k4 Is this still an issue with newest firmware? If not please close this issue.
[stale]

closed 2020-09-22 by sarf2k4