prusa3d/Prusa-Firmware #1194: MMU2 calibrate for different diameters of hobbed gears

bozzzzo, 2018-09-25:
I got a set of hobbed gears that were not all the same diameter (6.25, 5.89, 6.11, 6.26, 6.27) which made the second filament consistently fail to load because it was fed in about 2cm too less. (I'm getting sent a replacement set of gears)

The gears will never be exactly the same, so it would make sense to calibrate each position separately, not only the first.

The calibration could be entirely automated if filament sensor were used to detect when filament enters the extruder.

Probably related to some symptoms described in #1163

3d-gussner, 2018-09-25:
On the other side the load sequence could be automated as the filament sensor on the extruder should be able to sense the incoming filament being incoming, like autoloading. Only tricky could be to find the exact position at the bondtech gears so the tip forming performs as needed.

xiew, 2018-09-26:
This fix is a must. I can’t load anything other than my 1st filament, the rest fall short. Secret menu only calibrates the length of the first one. Please allow this calibration for each filament ASAP. Thanks!

bozzzzo, 2018-09-26:
@xiew the workaround is to put the smallest gear in position 1 and re-calibrate.

danielegger1, 2018-09-26:
Same issue here!
I can see that different positions moves farer into the extruder or not far enough

I will measure the diameter later today to see if I have also such differences in size

AbeFM, 2018-09-30:
Playing with swapping gears moves your problems from the first layer to the Nth layer - you grind notches in the filament, and then later you don't quite drive all the way to the bondtech's. Also you fill the whole printer with shreds of plastic.

I've gotten rid of the other issues, but constant grinding on half my channels in inherent in the way the system works right now.

KarlZeilhofer, 2018-10-16:
since MMU 2.0 firmware 1.0.1, I think we could close this issue! @bozzzzo can you please close it?

bozzzzo, 2018-10-16:
Installed 1.0.1 and could calibrate for all 5 filaments.

closed 2018-10-16 by bozzzzo