prusa3d/Prusa-Firmware #1196: MK3 3.4.0 Trying to exit Z adjust with a long-press prevents exiting at all

bozzzzo, 2018-09-25:
Turn on MK3. long press the button. Z adjust appears. long press the button again. cannot exit from the Z adjust anymore. probably related to #1195

stahlfabrik, 2018-10-08:
I have had the exact same issue happening on 3.4.1

Panayiotis-git, 2018-10-08:
Why I think that after adjusting the Z height, a simple press to the button exits the screen?

stahlfabrik, 2018-10-08:
@Panayiotis-git of course, if the bug is not in affect, that is how it SHOULD work.

When the bug happens, you can't leave the menu. You can only press reset.

Panayiotis-git, 2018-10-08:
Oh, I got it!

menu.cpp, line 121
void menu_submenu(menu_func_t submenu)

It should check if already in submenu then return.

Panayiotis-git, 2018-11-16:
So, new fix:
lcd.cpp, method void lcd_buttons_update(void), line 803
partially uncomment the else block:

           if (lcd_long_press_active == 0)
            { //button released before long press gets activated
                newbutton |= EN_C;
            }
            else lcd_quick_feedback();

maximlevitsky, 2018-12-22:
Still happens on 3.5.1. 100% reproducible.
Just do a long press to enter the Z axis mode and then another long press.

sgoll, 2019-03-07:
The same happens with recent firmware version 3.6.0.

It also happens when long-pressing the knob during a print. In that case, why does the Adjust Z Height show up anyway? Does this make sense, or will the next G-Code command just override any changes that I make? Should it be possible to enter Adjust Z Height only when the print is paused?

The printer does not react to any user commands until restarted, even when the Adjust Z Height dialog disappears eventually.

(In my case, I ended up there because I wanted to do Live Z Adjustment and wasn't sure how to reach that. I would have expected to go there by long-pressing during a print.)

ochm, 2019-03-17:
Yes, i found the same behavior.
When the button is long pressed during print, it displays Z level change.
Well, the level can't be changed, OK, so the next press should return display to main menu, but does nothing, and after some timeout, the display is returned to main menu, but from this time the button does not react on press, and no menu can be accessed.
IMHO during print the long press of button should start the live Z adjust,
and after return from this process (after short press or timeout) the whole display should be fully functional.
Thanks for repair.

Panayiotis-git, 2019-03-17:
The fix is a couple of posts above. You can give it a try. You have to compile the firmware yourself, though.

ochm, 2019-03-17:
Thanks for information.
I payed for whole printer incl. firmware, not for DYI.

myreauks, 2019-08-05:
This is still happening in the newest firmware. Even if you wouldn't add the live-z functionality during print for the long press, I think you should fix the printer user interface crash when long pressing the button during a print. I have a multiple hour print going currently, but cannot access the menu anymore as the printer is not responding to the button after long pressing it twice.

Is a fix coming in recent updates?

closed 2019-09-10 by DRracer

3d-gussner, 2020-07-13:
@bozzzzo Is this still an issue? If it has been fixed in a newer firmware version, please close this issue.
[stale]

commit 33037d36 (2019-09-10): Prevent LCD long-press re-entry (fixes #1196)