prusa3d/Prusa-Firmware #1204: Reinstate cooling fan OFF behaviour when M600 change or filament out.

Crowlord, 2018-09-27:
In the previous firmware the cooling fan switched off when filament was changed. This was a great improvement as the extruded filament went neatly onto the bed and didn't get blown all over the buildplate (and sometimes even stuck itself in a neat little birdpoo pile)!

This has now regressed in 3.4.0 to the original behaviour. 

Can we have it back or was there a good reason for its removal?

gudnimg, 2022-12-10:
Whether this was ever fixed between now and when this ticket was created is unknown to me. But issue was present in FW 3.11 and will be fixed in 3.13 via https://github.com/prusa3d/Prusa-Firmware/pull/3746

@Prusa-Support can we close this ticket in favor of more recent https://github.com/prusa3d/Prusa-Firmware/issues/2717 ?

Prusa-Support, 2023-01-03:
Thank you.
This issue will be closed in the same way as issue https://github.com/prusa3d/Prusa-Firmware/issues/3670 that led to such Pull Request.

Michele Moramarco
Prusa Research

closed 2023-01-03 by Prusa-Support