prusa3d/Prusa-Firmware #1207: after upgrading to 3.4.0 unable to move the z axis from lcd

uepsie, 2018-09-27:
After I upgraded to 3.4.0 (which by the way doesn't flash properly all the time) my printer refuses to move the z axis in the menu after long-press the knob.
The values change on knob rotation and there is a gentle buzzing from the stepper(drivers) but nothing moves.
Controlling the z-axis via terminal or octoprint works fine but also does not work after trying to change the position from the lcd

stahlfabrik, 2018-10-08:
I ran into a similar issue: I could move Z by entering the "long press" menu - but I could not exit the menu again - I do not know if it is directly related - but the point of failure in that menu - that seems very near. I used 3.4.1.

stahlfabrik, 2018-10-08:
Nah - seems my issue is more exactly the same as #1196 

But nevertheless, that menu item seems still buggy

Prusa-Support, 2022-05-11:
Hello! I am absolutely sorry we haven't replied to this for so long. I assume this has been solved after installing newer firmware, right? In case it's still relevant, please let me know, we can look into it. 

Alessandro Pantaleo
Prusa Research

closed 2023-09-13 by github-actions[bot]