prusa3d/Prusa-Firmware #1208: Speed change returns to 100

jonbet83, 2018-09-28:
After a filament change on MMU 2 the speed defaults back to 100 even whens its been changed during the print.

mikh3x4, 2018-11-21:
I would like to confirm this is indeed true and fixing it would make my workflow so much easier

TickTock-, 2019-04-07:
Also, extruder temp reverts back to original temperature next time that same filament is loaded during the print.  Probably flow and the rest of the tune settings...  Tune needs to store independent values for each of the five MMU sources.

JohnnyDeer, 2021-01-20:
Speed resetting between filament changes has already been fixed, I'm closing this issue.

closed 2021-01-20 by JohnnyDeer

polygonalenippel, 2021-01-31:
I am experiencing speed reset between filament changes as well with the latest firmware.

- mk3s+, fw 3.9.3 3556
- mmu2s, fw 1.0.6

simonemarin, 2021-04-17:
> I am experiencing speed reset between filament changes as well with the latest firmware.
> 
>     * mk3s+, fw 3.9.3 3556
> 
>     * mmu2s, fw 1.0.6 372

Same version same issue here