prusa3d/Prusa-Firmware #1209: New Filamnet loading idea

jonbet83, 2018-09-28:
On MMU 2 when a new filament is loaded the bond tech gears aren't moving and this in turn can cause girding as each filament path is a bit different due to gears having different diameters.

so what cant they be moving at the same rate as the MMU loader to help draw the filament into the hot end rather than create a stopping point. it would be much more fluid and the gear diameter wouldn't be so much of an issue either would it?

3d-gussner, 2018-09-30:
Good idea, only downside is that the printer wouldn't know when the ramming has to be done while unloading.

jonbet83, 2018-09-30:
It’s more of an loading issue than unloading I’ve found, after it’s been printing the retraction length is always the same as the filament is at the hot end for cooling moves etc, i don’t think position of the filament loading would be so critical if the gears were moving at the same rate as long as the length is just long enough to feed Into the bondtech gears, it’s over the tower anyway to take up any over or under extrusion and prime ready to move to the print. Feeding filament into stopped gears is like driving a car into a wall at speed.

jonbet83, 2018-09-30:
I know there saying a new firmware will be out to do each length but it’s all pointless really, if the gears moved none of that would be needed

AbeFM, 2018-09-30:
I believe the gears ARE running, just slowly. Peter on the forums suggested upping their speed which is next on my todo list.

You'd think you could monitor the current on the Bondtechs to get some information out about when you made contact.

JohnnyDeer, 2021-01-18:
This problem is solved by MMU2S, I'm closing this issue.

closed 2021-01-18 by JohnnyDeer