prusa3d/Prusa-Firmware #1213: Automatically sense filament length on MMU 2.0

TheDuggem, 2018-09-30:
Can the drivers in the MMU be used to sense when the filament hits the Bondtech gears in the extruder similar to the way they are used to detect the end stops?

3d-gussner, 2018-09-30:
Good idea, here some alternatives:
- After calibrating the 1st extruder as described the MMU2 could load the following filament and as soon the FINDA triggers it will push 100mm (calculated from 1st extruder) and then pull back until FINDA stops triggering. this way it could calculate the difference from 1st to following extruder and store the values in the eeprom.
- Calibrating all extruders manually
- Use the MK3 filament sensor to sens if the filament got in the filament. Should work as Autoload function

AbeFM, 2018-09-30:
It seems like this is something that could be done iteratively - even using the lower bondtech pair to trigger by watching when it's getting fought by the MMU's drive.

Even with an unreliable sensor, you'd think 10 repeats would do it - certainly if the user can watch it happening?

TheDuggem, 2019-02-15:
This is no longer relevant because of the new MK3S/MMU2S filament sensor design.

closed 2019-02-15 by TheDuggem

AbeFM, 2019-02-15:
You mean Prusa's indirect filament sensing?

I don't see how that negates the need for this feature. At best, it says some people will need it more than they did before because now their filament sensor might actually work.

And certainly anyone has *A* distance from sensor to Bondtech's, and many like me are running a custom set up and still need access to that number.

I've spent a fair amount of time trying to get the older ZB firmware to handle that distance right, it is sensitive to what you pick.

So I don't see why this thread is closed.