prusa3d/Prusa-Firmware #1214: Release a full Menu Tree for MMU

AbeFM, 2018-09-30:
There are 3 buttons on the MMU, some of them do predictable things, some times.

There seems to be distinct modes of flashing lights and solid lights, and in various states, the button functions differ.

Often I think of changes, but frankly, I'm lost as to what's going on most of the time.

A user friendly state diagram that says how to recognize a state from the lighting, and a description of what each button does when tapped, pressed, or held, would be a lot easier to deal with.

 If there is something like that,and I'm missing it, could you point me there?

mylife4aiurr, 2018-10-01:
This isn't quite what your looking for. But take a look at these links. They explain the lights and buttons. Some of the buttons functions change depending on the jam direction which can be confusing at first.

https://manual.prusa3d.com/Guide/Status+LED+indications+explained/762?lang=en

https://manual.prusa3d.com/Guide/MMU2+filament+jam/779?lang=en

KarlZeilhofer, 2018-10-16:
since this is a duplicate, we should close it.