prusa3d/Prusa-Firmware #1215: Semi-Automated Filament Trimming, feature request

AbeFM, 2018-09-30:
At the end of most prints - especially in development where most of them are failing - I'm left with stringy ends on my filaments.

As most aren't pinched, I can shove 3 of them through, generally the "selected" channel as well, but the one next to the active channel is pinched and you have to force it through.

When you're done, you get this:

[image]

Which is super easy to work on! It probably accomplishes what the build in razor was supposed to do, and it seems like something that could be added to the hardware - spitting out 3 strands, letting you trim them, then 'loading' again might work - or doing them one at a time with the selector in play might make more sense - unscrew it as shown.

Does this make any sense? I have the most luck when I start with 5 nicely tipped/trimmed just loaded filaments, perhaps it's worth doing.

KarlZeilhofer, 2018-10-16:
There are 2 options: 
1. I think the most easiest way would be an option in the menu called "inspect filament tips", which will push every filament 5cm out of the bowden socket (wich tube disconnected, of course). Then the MMU can retract it knowing the exact length of the filament tip. 
2. the selector will be moved into parking position to the very right and every filament is pushed out for about 10cm. The user can then move one filament after the other with the user buttons on the MMU, to put them back in place: Middle = retract filament with low speed, Left/Right: select filament with idler drum.

AbeFM, 2018-10-17:
That's pretty much what I was thinking. 
1) Move selector to far right, which triggers a home of everything (you're not using anything, so why not?)
2) If you (whatever, both buttons, or something) it triggers this shoving everything out, pulling everything back. You do run the risk of pulling it back too far and having to feed it from the far end.

JohnnyDeer, 2021-01-13:
Hi, since FW 3.9.0 is MMU2S selector blade activated, so I'm closing this issue.

closed 2021-01-13 by JohnnyDeer