prusa3d/Prusa-Firmware #1217: Can not compile with #define TMC2130_DEBUG

mat100, 2018-10-01:
Firmware can not be compiled with flag TMC2130_DEBUG enabled. 

Compiler reports an error:
"invalid conversion from 'void (*)(uint8_t) {aka void (*)(unsigned char)}' to 'menu_func_t {aka void (*)()}' [-fpermissive]"

Please help

LowM4N, 2018-10-10:
With which version of Arduino IDE you tried to compile it? In the README stays you should try it with older versions like 1.6.9, but if you are on MacOs than you should use 1.6.13, cause the 1.6.9 have problems with MacOs High Sierra.

On my MacBook I can compile it with 1.6.13 without any errors.

mat100, 2018-10-11:
Hi,

I tried it in recommended version 1.6.9 with this error. I am now trying the version 1.6.13 and compile is without error.

Thank you for your help

closed 2018-10-11 by mat100