prusa3d/Prusa-Firmware #1225: FlashAir: firmwire is unable to detect IP address

crazyquark, 2018-10-04:
I have exactly the same flash air version as the one in the tutorial on the blog.
I can find the card's IP address via my router's logs but it does not show up in the Support menu.
I am running FW 3.4.0 but I will update to 3.4.1 today.

crazyquark, 2018-10-04:
It's the same on 3.4.1

ironjoe13, 2018-10-05:
What size is the card? There have been instances that users reported that the 64GB version does not work, but the 32GB works fine. I use the W-04 version on MK3 and MK2 and it works just fine for me- however I do not need to se what the IP is after reserving that IP on my router.

crazyquark, 2018-10-06:
It's a W-03 32GB card, I just upgraded it to the latest available firmware from Toshiba and the problem persists.
Yeah, sure it's not a BIG issue but maybe it's a fixable one. If need be I can provide some debug logs if you let me know who to do that(I suspect some SDIO command fails?)

Saij, 2018-10-16:
I have the same problem. Card is a W-03 32GB and in the support menu there is no IP. But I can ping it and access it through HTTP and WebDAV. Card Firmware is FA9CAW3AW3.00.02 and printer firmware is 3.4.1-rc1-1354

stephanr, 2018-10-28:
Same here on MK2.5 with Firmware 3.4.1-1356
The Card is an W-03 with 8GB and Firmware FA9CAW3AW3.00.02
According to the Toshiba Website the Firmware V3.00.02 from April 24, 2017 is the latest available Version.
The Card can be used without Problem but the IP-Address is not shown in the Support menu.

nikst4, 2018-11-05:
Same here.
Toshiba Card W-04 with 32 GB.
Worked with firmware 3.4.0, now quit workling.

ptr727, 2019-02-13:
Similar problem.
W-04 64GB, currently the only model available on US Amazon.

Card comes formatted exFAT, Mk3 does not read exFAT, would be nice if you support exFAT.

I updated the card firmware.
I re-partitioned and reformatted the card to 31GB FAT32.
I reinitialized the card.
I edited the config to put it in STA mode.
The Mk3 can now read the SD card, and print files from the card.
I can connect over WebDAV and copy files to the CD card.

But, the Mk3 is not recognizing it as a FlashAir card, just a regular SD card.
After copying files to the card, I need to remove and re-insert the card, else the Mk3 does not see the newly copied files on the card.

nikst4, 2019-02-13:
Thank you for sharing this info!

In my opinion new firmware should adress this problem!

closed 2023-09-13 by github-actions[bot]