prusa3d/Prusa-Firmware #1226: MMU2 - single mode issues with Octoprint

Petclaud, 2018-10-04:
When using Single Mode and sending GCODE via Octoprint to my MK3 with MMU2, the printer starts printing directly with the second filament and does not ask on its display which filament to use. Also the filament loading is not done correctly, since although the printer head makes the path of a purge line and the extruder motor is rotating, nothing is being extruded since the filament is initially not loaded far enough into the extruder.

Maybe this is already reported somewhere but I could not find it. Apologies if reported double. 
On the forum this phenomena is also described here: https://shop.prusa3d.com/forum/assembly-and-first-prints-troubleshooting-f54/octoprint-mmu2-t24543.html

PavelSindler, 2018-10-05:
Thank you for reporting this issue.

KarlZeilhofer, 2018-10-08:
I can confirm the report, just had the same behaviour. I expected it to print with the 1st filament. I think, it was the last time I used the "single filament mode".

szabozsm, 2018-10-11:
I can confirm this. The single filament mode doesn't work through Octoprint. But the same gcode works when printed from SD card.
(Firmware 3.4.0 and 3.4.1)

milindur, 2018-10-13:
Same here with FW 3.4.1/1.0.1. When loading the same gcode from SD-card the printer asks after heating and calibration for the filament to use. When using OctoPrint, the print just starts with filament 2.

TheFitzZZ, 2019-05-05:
Pretty sure this can be closed. Printer asks for filament number in single mode printed through OctoPrint.

nunofgs, 2019-05-22:
Is there any way to specify the filament number beforehand, or answer through Octoprint itself?

This is making me go to the garage every time I want to start a print :(

Vortec4800, 2019-06-24:
I'm having this issue too. It would be nice to have a gcode command to select a filament number when starting from USB so Octoprint can "answer" the question.

closed 2023-09-14 by github-actions[bot]