prusa3d/Prusa-Firmware #1232: Octoprint wont connect to mk2.5 FIRMWARE 3.4.1

BotleHewer, 2018-10-05:
I could not get my octopi to connect with 3.4.1, I just flashed 3.4.0 and it connected right away.

Prusa mk2.5
Octopi on RPi2

Tech-outreach, 2018-10-06:
had the same issue, Octoprint . Can't find the baudrate with 3.4.1, I reinstalled 3.4.0 and had no issues with Octoprint

sky4055, 2018-10-06:
same issue here, i downgrade to 3.4.0 and had no issues.

tibahut, 2018-10-06:
Same issue with PC USB connexion. A fallback to 3.4.0 firmware makes it work again.

pet3rs, 2018-10-07:
same here. downgrade to 3.4.0 fixed it.

KarlZeilhofer, 2018-10-08:
On the MK3 (with MMU2) it works without issues. Connected to a PC running Debian.

Picazsoo, 2018-10-08:
Same issue on my mk2.5 - had to downgrade to 3.4.0

ram64, 2018-10-11:
I am experiencing the same issue with firmware version 3.4.1 on my MK2.5. The printer resets when OctoPrint tries to connect but OctoPrint shows as not connected.

asterixorobelix, 2018-10-11:
I can chip in here too - I am experiencing the same problem with Octoprint and firmware 3.4.1
I have even downgraded the firmware all the way to 3.1.1 and the problem still occurs.
See my issue on Octoprint's github: 
https://github.com/foosel/OctoPrint/issues/2843#issuecomment-428698447

owey71, 2018-10-15:
I set baudrate to AUTO and managed to connect to lower baudrate(57600). Higher than that won't connect.

Tchoubaka, 2018-10-17:
I just finish my upgrade from MK2S to MK2.5 and install the latest driver. Then I was able to connect via USB and flash the firmware to the latest version 3.4.1-1356. Now I cannot connect to my printer via USB with Slic3r Prusa 1.41.0, always get the error: connection failed. Event if a see some action (character change for square when the connection start) on the display of the printer. I try to change the speed to 57600 but did not work. I will try to downgrade the firmware to 3.4.0.

AMFkuna, 2018-11-11:
Strange thing,
I have same issue, but..
I have enabled filament sensor and baudrate set to AUTO, printer connects after some time.
If I disable filament sensor and set baudrate to AUTO, printer not connects and cycling with trying baudrates.
If I set baudrate to 115200 and have disabled filament sensor, printer connects immediately.

foosel, 2018-11-15:
@yschroeder there's actually a third way: The firmware does what it should do and sends a `start` when it has initialized its serial line. That will make OctoPrint try another `M110`. 

Why does OctoPrint send an `M110` right on connect anyhow? Because there are printer controllers out there that *don't* reboot on serial connect, which also don't send anything on serial connect that would allow communication to get going, so OctoPrint has to do the first step and tries to perform a handshake in case it's connected to such a machine. If it isn't, that machine will (usually...) then send a `start` and all that wall of text on boot up anyhow, and OctoPrint will realize "oh, look at that, it's a printer and it just started, let's try the handshake again". There's even the existing option to have OctoPrint wait until it sees that `start` before it even sends *anything* (because some controllers really don't like it when they get any data before they are fully there yet). 

But without this firmware here sending this `start` message even though it's restarting, it's making things tricky. I honestly don't want to have to add exceptions and special settings for various firmware variants out there all the time to work around their specific quirks, especially if new quirks get added through firmware updates constantly. That simply doesn't scale and becomes a horrible mess to maintain.

DRracer, 2020-09-22:
Is this issue present in the latest FW 3.9.1? If not, we'd like to close this issue.

AMFkuna, 2020-09-22:
No this issue NOT present in 3.9.1.

closed 2020-09-22 by DRracer

rubin110, 2022-11-08:
Seeing this issue (or something similar) with 3.11.0. Selecting `AUTO` for the port fixes the issue.