prusa3d/Prusa-Firmware #1233: Inefficient printing when sections of parts long way from each other...

AikiAviator, 2018-10-06:
I have created a new stl file in FreeCAD and transferred it to Slic3r 1.41.0+64Bit.

This file has two vertical sections at either end of the part. The printing does one layer at a time beginning at the same "end" of the part. Basically, it returns to the same position before commencing next layer.

This is very inefficient as if it could commence the next layer at the point of the last layer it would reduce the travel time for each layer by half.

The layer height that this commences is 5.2mm and continues up to 48mm. If each layer travel time is 2 seconds then the total savings for a part like this would be (42*10*2 seconds) = 14 minutes on a print of 2 hr 34 mins. A savings of ~14%. 

Basically if there was a shortest point calculation built into the gcode generation, this could have major impacts on the print times.

I have attached the STL file and the gcode generated for reference.

PS: I am fairly new at this so if there is already a setting to address this please advise.

Regards,
Andrew

GCode
[https://www.dropbox.com/s/i1dm82l1roe8gbt/Fridge%20Filter%20Holder_0.2mm_PLA_MK3.gcode?dl=0](url)

STL
[https://www.dropbox.com/s/ra5tgnj94429rpc/Fridge%20Filter%20Holder.stl?dl=0](url)

tarikku, 2018-10-06:
Hi Andrew, and welcome!
An interesting suggestion, I wonder whether there is some good non-obvious reason for this behavior.

However, you might want to open an issue on this topic over at the prusa3d/slic3r Github, as such movement patterns will have been specified by the slicer (I.e. present in the gcode file), and not by the firmware itself. You can visualize this in Slic3r by looking at the preview, I think it is visible only after exporting the gcode.

The firmware just follows the movement commands present in the gcode and might do small changes to the way the filament is extruded etc, but does not decide itself where the extruded assembly will move mid-print, at least not during an error-free print.

BR,
Tarik

AikiAviator, 2018-10-07:
Yes, had nearest on and have logged in Prusa thread.  Thanks.

closed 2023-09-06 by github-actions[bot]