prusa3d/Prusa-Firmware #1236: FW 3.4.0  loses settings after PAUSE

GWdd, 2018-10-06:
During my latest print I have been pausing frequently to correct minor part support issues.

On a couple of occasions when RESUMEing, the printer would indicate it was heating the hotend but the setpoint was 0c. I went to the PREHEAT menu and manually preheated; after the hotend reached the preheat setting, the printer would resume printing ... whew! This happened a few times.

On another pause when I resumed the print I noticed some sagging in places that hadn't been sagging, and discovered the filament fan was OFF. I manually turned it back ON at 100%. This happened only once. 

The .gcode file is about 158M. This was a LONG print: 83 hours. Many pauses, some lasting 30 minutes to allow adhesive to dry (Elmer's Glue supports for the supports that had detached).

onverra, 2018-10-08:
I had the same behavior twice (temperature target at 0C, fan OFF), while printing smaller prints than GWdd. It happened for a 25 hours print to me.

PavelSindler, 2018-10-19:
We have new pause print with immediate function coming in next release, so this will be also solved there.

closed 2022-05-11 by Prusa-Support