prusa3d/Prusa-Firmware #1241: MMU2: confusing term "Load Filament"

KarlZeilhofer, 2018-10-08:
that term is very confusing, since it is used for loading the extruder with filament and for loading the MMU2 with filament. For unloading there is a different term (unload vs. eject), but I'm not happy with them, since they are not self explaining. 

For loading I would suggest: 
* Load Filament into MMU
* Load Filament into Extruder

AbeFM, 2018-10-10:
I am becoming partial to the term "Preload".

laird, 2018-10-11:
It's an important distinction, but the display is small, so how about compressing down to:

- Load Extruder
- Load MMU

AbeFM, 2018-10-11:
I like it

inquisitiveKangaroo, 2018-11-21:
Solved in 3.5.0 - RC1 by 'Load filament' / 'Load to nozzle' features. Thank you for feedback, we appreciate it.)

closed 2018-11-21 by inquisitiveKangaroo