prusa3d/Prusa-Firmware #1250: No Z High shown on Display just a ?

galluslongus, 2018-10-10:
I compiled 3.4.1 on my own cause of PWM_MOTOR_CURRENT Settings. I use different Motors.
After Flashing new FW Z High on the Display is shown just the "?".
I did flashing compiled FW from Prusa´s website and theres all like it should.
I just edited the configuration_prusa.h for PWM_MOTOR_CURRENT settings, nothing else.
Anyone an idea?

Thanks
Michael

LowM4N, 2018-10-10:
I have the same issue. I cloned the MK3 branch, adjusted the homing currents and stall guard thresholds, because I got 3 of this [steppers](https://www.omc-stepperonline.com/nema-17-bipolar-45ncm-64ozin-2a-42x40mm-4wires-w-1m-cable-and-connector-17hs16-2004s1.html) for x/y/e and they didn't home properly with the default settings.
After a G28 is see only a '?' at the z-positon on the LCD.
When I try to run the first-layer calibration, than I see only a '-' in the z-offset.

Barnabasek, 2018-10-10:
See #1117

galluslongus, 2018-10-11:
I have 7 of the platform.txt on my system. Its a macbook pro / High Sierra.
I pasted the line in all these files BEFORE i opened this post with no luck.
Is it a hidden file?

Panayiotis-git, 2018-10-11:
The correct file is the one referenced at the https://github.com/prusa3d/Prusa-Firmware/issues/1117:
hardware/arduino/avr
The file already contains the property and you should change the value.

If the problem still exists after the change, consider reinstalling the arduino IDE (prefer the portable version).

LowM4N, 2018-10-11:
@galluslongus create a '~/Library/Arduino15/packages/rambo/hardware/avr/1.0.1/platform.local.txt' with this as content

```
compiler.c.elf.flags=-w -Os -Wl,-u,vfprintf -lprintf_flt -lm -Wl,--gc-sections
```

galluslongus, 2018-10-11:
Awesome, that platform.local.txt did it. Thanks a lot

closed 2018-10-11 by galluslongus