prusa3d/Prusa-Firmware #1254: compile 3.4.1-1356 question mark in Z height

Strudsen, 2018-10-12:
I was trying to compile the 3.4.1 firmware for my MK3 with mmu 2.0, I have just taken the souce for in releases, and I have copyed the 1_75mm_MK3-EINSy10a-E3Dv6full.h from variants, and I have successful compiled it with Arduino 1.6.13 for mac.

but when I see on the display, it says Z ?, and when I try to extrude, it also says ? in E.

Do I miss a step here ?

the reason for the custom firmware is that I want to mount a Bondtech BMG.
[image]
[image]

Panayiotis-git, 2018-10-12:
Check the closed issues. You are going to find a dozen times exactly the same question.

Prusa should mark with bold the critical parts of the build instructions.

https://github.com/prusa3d/Prusa-Firmware
Step 1.3

Barnabasek, 2018-10-12:
Please read #1250 and you will find a solution. Please close this issue.

closed 2018-10-12 by Strudsen