prusa3d/Prusa-Firmware #1255: can't compile current Firmware

Devilscave, 2018-10-13:
I always get the following error while compiling the current firmware.
I have all done as described in the manual, I also use the IDE 1.6.9. But it always brings the same mistake. On the firmware, I have not changed anything, had anyone ever?

g:/3d drucke/mk3 software/arduino-1.6.9/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr6\libc.a(isspace.o): In function `isspace':

(.text.avr-libc+0x2): relocation truncated to fit: R_AVR_13_PCREL against symbol `__ctype_isfalse' defined in .text.avr-libc section in g:/f platte/3d drucke/mk3 software/arduino-1.6.9/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr6\libc.a(cty_isfalse.o)

collect2.exe: error: ld returned 1 exit status

exit status 1
Fehler beim Kompilieren für das Board RAMBo.

Panayiotis-git, 2018-10-18:
This is a place to post issues regarding the firmware. Not a place to post the failed tries to compile the firmware.
Please, have a quick look at the closed issues. I have lost the count of the issues opened regarding the exact same thing.
You may consider to use the forums to get support on this issue.

3d-gussner, 2019-02-16:
Checkout https://github.com/3d-gussner/Prusa-Firmware/wiki/How-to:-Build-Prusa-Firmware

cimoalpacino, 2019-07-21:
this can also be closed

closed 2019-10-08 by Devilscave