prusa3d/Prusa-Firmware #1256: document mk3 firmware build procedure / official build are not reproducible

jhoblitt, 2018-10-14:
I've made a local build of `v3.4.1` with `1_75mm_MK3-EINSy10a-E3Dv6full.h` copied to `Configuration_prusa.h` without modification. Using ardunio ide `1.6.13` (as `1.6.9` is broken), with the recommended `platform.txt` directive and the `1.0.1` "RAMBo" board, per the `README.md`  However, the size of an exported hex dump is significantly smaller than the official 3.4.1 firmware.

```
$ ll prusa3d_fw_MK3_3_4_1.hex
-rw-rw-r-- 1 jhoblitt jhoblitt 848642 Oct  4 19:49 prusa3d_fw_MK3_3_4_1.hex
$ ll Firmware.ino.rambo.hex 
-rw-rw-r-- 1 jhoblitt jhoblitt 686949 Oct 13 21:28 Firmware.ino.rambo.hex
```
Which means that the instructions in the current `README.md` are insufficient to reproduce the official builds.

Devilscave, 2018-10-14:
I can not even reproduce them because I get a bug in the IDE.

Devilscave, 2018-10-16:
How and which "platform.txt" did you change? I had big difficulties. Now it works well with the compiling but the size is also very different, about 200kb. Why is your 1.6.9 IDE broken?

Saij, 2018-10-18:
Maybe you are missing the languages in your hex file?

trevjonez, 2018-10-20:
you might give arduino-cli a try. I was able to get it working and find it much nicer than the IDE. at some point I want to see if I can work the arduino-cli into a cmake config so that I can get CLion to play nice with the project.

main thing to do with the arduino-cli is `.cli-config.yml` to configure the board manager additional_urls.

jhoblitt, 2018-10-20:
To be more pedantic, I am able to build and install a working firmware.  However, it is disconcerting that there is such a large size difference between a self built hex dump and the official published releases.
 
Obviously, there are uncommited changes to `Configuration*.h` that are made for an official release so it doesn't identify as a dev build along with some other source modification or build tooling changes that greatly inflate the size of resulting hex dump.  It seems probable that release process notes and/or scripting already exist that aren't committed to this repo.

trevjonez, 2018-10-20:
another huge indicator to me is that there is at least one and likely more IFDEF keys that are not defined anywhere. commented or otherwise. smells like an external cmake file might be at play.

cadriel, 2018-11-11:
I also have this issue. I have the project building `v3.4.1` in both a mac and windows environment - both produce a file size of around;
```
$ ls *.hex
-rw-r--r-- 1 me staff 682866 Nov 11 12:37 Firmware.ino.rambo.hex
```

I'm slightly hesitant to flash this without knowing it'll work given the file size differences between my firmware and the official distributed one.

Has anyone tried flashing their's?

3d-gussner, 2018-11-11:
Prusa MK3 firmware source code has multiple languages disabled by default now. The release hex files include more languages which have been added via few other scripts. To be found under the Lang folder.

3d-gussner, 2018-11-12:
As far I can say the tagged code used by Prusa as a base for compiling. I am not 100% sure what their procedure is but here my thoughts:
1. Copy one variants/???.h file as Configuration_prusa.h in Firmware folder
2. They increase the build number min by one (comparing source code build number vs. LCD support info)
3. Change the fw_debug from unknown to GOLD or RC (again source vs. LCD support info)
4. Change the repository from unknown to Prus3d something (again compared)
5. Change the config.h value to multiple languages
6. Compile the code (with modified build.path in the arduino preferences.txt)
7. Run the '/Lang/Lang-build.sh all' script to generate needed files
8. Run the '/Lang/re-build.sh' script to merge the generated firmware hex file with the multiple language files
9. Post process to new filename
10. Publish the release hex file

For MK2.5 the procedure will be bit different at step 7 as the MK2.5 firmware just gets two languages: en+de, en+it, en+es, and so on

You can search the issues for more information. I updated one of my issues yesterday as the languages part has been fix-bugged and I can't get the right results anymore.

It is a pity that still don't have a official how-to-compile from source code to final hex file instructions.
I spend hours to figure out how it done.

3d-gussner, 2019-02-16:
Checkout https://github.com/3d-gussner/Prusa-Firmware/wiki/How-to:-Build-Prusa-Firmware

closed 2022-12-29 by wavexx