prusa3d/Prusa-Firmware #1259: MK3 3.4.x software caused layer shift

hon-po, 2018-10-15:
Just replaced the X & Y grooved rods, and no problem with the X & Y motion (this is not my first printer).
Large prints failed consistently after a few layers.

Post with picture & link in here:
https://shop.prusa3d.com/forum/hardware-firmware-and-software-help-f64/firmware-3-4-x-bug-layer-shift--t25199.html

hon-po, 2018-10-16:
This is caused by filament sensor, the post linked above is updated with more information.

It is more usable to have a LCD setting to enable filament sensor for run-out only detection, ignoring filament slippage.

Prusa-Support, 2022-05-24:
Hello! Sorry for intervening on this so late. Was the issue still relevant? I believe it should have been solved with the more recent firmware versions released in the meantime. In case of current issues with this aspect, do not hesitate to let me know via email info@prusa3d.com mentioning me ("attn: Alessandro"), we'll look into it. Closing the issue here for now. 

Alessandro Pantaleo
Prusa Research

closed 2022-05-24 by Prusa-Support