prusa3d/Prusa-Firmware #1260: [3.4.1] Selftest failing with Y axis Length error

nicolaszundel, 2018-10-15:
Hi,

I'm running 3.4.1 and when I run the selftest, the test fails with Y axis length error.
When measuring Y axis length, in one direction when it reaches the end it just goes back to other direction but when reaching other end it continues instead of returning back it continues and crashes then the error message is raised...
I tried many time and only one out of 10 it worked... So I flashed back firmware 3.4.0 and could run the selftest without issue. I ran then the calibration wizard  which went fine.
After this I upgraded the 3.4.1 and could run the selftest without issue.
Does anyone else have this issue ?

Crunch69, 2018-10-17:
yes I too experienced this exact issue.  when doing the selftest in fw 3.4.1 the y axis crashes bangs the bearings hard into the y axis rod holders.  Customer support told me it was hardware and I should not run the FW 3.4.0

Prusa-Support, 2022-05-10:
Hello! I am sorry for the late response. Since it was long ago, and there have been many releases in the meantime, I believe it is all resolved, likely with the successive release, so closing the issue based on this - in case of anything still related with FW-related self test fails, please let me know via email info@prusa3d.com mentioning "attn: Alessandro" in the subject, I'll help with it. 

Alessandro Pantaleo
Prusa Research

closed 2022-05-10 by Prusa-Support