prusa3d/Prusa-Firmware #1261: Printer will not do the same thing twice in a row

AbeFM, 2018-10-15:
This is a pretty simple request:

I would like to turn the printer on, select the first file from the SD card, and have the printer DO THE SAME THING it does if I do it again.

When I turn the printer on cold, with all the filaments pre-loaded, and the SD card untouched, the MMU may or may not home the drum. It may or may not home the selector. The selector may end up at the far right, the far left, or somewhere in the middle.

When the printer is booted, the MMU will do this again. Perhaps moving, perhaps just flashing lights.

It will likely feed some filament out, possibly into the waiting selector, generally not. It might ask me to trim the end, and if you press any button at this point, it will bend the extended filament over and jam. Maybe. It might feed the filament forward, and it might feed it backward.

Something happened with the newest firmware - I'm not sure if it's 1.0.0.-117 on the MMU or 1.3.4-1356 on the printer... But am important assumption of g-code is that following the instructions in order will leave the printer in a defined state.

I have asked around a few times, and I'm looking for an answer: How do I get the printer to do the SAME thing with the SAME inputs? Trouble shooting is getting impossible when the printer literally is RANDOM.

I've tried posting videos and there's no responses - it is release firmware, so I am to assume this is the way the device is INTENDED to function.

inquisitiveKangaroo, 2018-10-17:
Hi,
unfortunatly I wasn't able to reproduce such behaviour. In my case the MMU is acting the same everytime. Could you please link me some of those videos? And reassure yourself having newest firmware version which is 3.4.1-1356 for printer board (mk3/mk2.5) and 1.0.1-126 for MMU2 unit (12. 10. 2018). 
https://www.prusa3d.com/drivers/

Wrong selector homing position is most often caused by fragment of filament stucked near the razor. When all lights are flashing means the finda indicates a filament (or some mess) inside the selector.

Please make sure none of those unpleasantness occurred. We would be grateful if you shared us more specific informations such as a video of your printer's behaviour with each move and result step by step.

I hope we will solve your ticket as soon as possible and thank you for your feedback.)

AbeFM, 2018-10-17:
First off: thanks for caring if our printers work! 

Old video from the tenth: https://photos.app.goo.gl/1pR6eYef3yu8bTtd8

It's fairly representative. It's on -117 firmware.

https://photos.app.goo.gl/HFaSKTjqizMe6Q4k9
This is from Monday. After ~3 resets, I started filming. The video shows one reset, one me-missing-the-button, another failed reset and then a successful one.

I'll get the new firmware - I've asked several times if there's something new so I'm glad to get a response.

While I have your ear: is the printer firmware packed alone different from the firmware packaged with the MMU firmware? They have the same number and file size.

AbeFM, 2018-10-17:
So far it makes it through the reset every time. The drum always twitches but sometimes I don't think it really homes.

Sometimes a reset will leave it almost homed - only the selector will stay to the far right instead of moving to the first position. When this happens, you cannot move the selector with the buttons, although the light moves around as if the unit is functioning.

Video of such: https://photos.app.goo.gl/Vsk9NhBDgp6kX1HR8

The drum has lost track of where it was a couple times (due to larger force of a partially fed and bent filament at the rollers.

When doing a "load all", it can get lost, and spit filament out for at least a meter. I believe it used to hunt after not seeing a signal, but that behavior's changed I think.

In short, it's working MORE, but not working.

AbeFM, 2018-10-20:
>And reassure yourself having newest firmware version which is 3.4.1-1356 for printer board (mk3/mk2.5) and 1.0.1-126 for MMU2 unit (12. 10. 2018).
Done

> Please make sure none of those unpleasantness occurred.
Done

> We would be grateful if you shared us more specific informations such as a video of your printer's behaviour with each move and result step by step.
Done

> I hope we will solve your ticket as soon as possible and thank you for your feedback.)
Could you give ME feedback i I gave you what you needed or not?

closed 2023-09-14 by github-actions[bot]