prusa3d/Prusa-Firmware #1265: On MMU-Single prints, ask user for material as soon as print is started.

AbeFM, 2018-10-18:
My third post about this was over here, and now I've been kicked to you guys?
https://github.com/prusa3d/Slic3r/issues/1333#issuecomment-430900510

As a general suggestion. instead of telling the user to post somewhere else - which will likely be done in an amateurish way, couldn't one of the developers post it in the right place and link to it, so we can continue to be maximally helpful.

The current system of 4 or 5 people telling me it's not THEIR specific fault misses the point: We're posting on here to communicate to the development TEAM things which might help. Help us to help you.

ruedli, 2018-10-20:
The gcode that triggers the prompt is T?. It is in the Slic3r "custom"gcode for the start of a print.

When issuing the same gcode through USB, like through Octoprint it is even more problematic, as it does not prompt, but will "select filament" in channel 2. If that one is empty (I use channel 1 normally) you are prompted to load filament in channel 2.

I agree with AbeFM, the prompt and _selection_ should be earlier. Then, this provided value should be stored and used when the filament can be loaded, like when the extruder is at the configured temperature,

In my opinion, selecting the filament (a user action) and loading the filament (a machine action, when it can be loaded) should be separated actions / gcodes. Combining the two is confusing, as they are very different things.

inquisitiveKangaroo, 2018-10-22:
Hi,
Thank you for feedback. We already noticed this delay may be annoying and we are working on improvement. Actually the solution is not trivial so we would like to solve it with best possible practice. I believe it should be implicated in the next release. 
We appreciate your effort to help us improve.
Happy printing.)

inquisitiveKangaroo, 2018-11-21:
Solved in 3.5.0 RC1

closed 2018-11-21 by inquisitiveKangaroo