prusa3d/Prusa-Firmware #1266: Extremely slow print with OctoPrint

dikodahan, 2018-10-19:
I am using the latest PrusaPrint image, fully updated to the latest version. This is along with the latest v3.4.1 firmware on my MK3 and the latest 1.41.1 RC build of Slic3r.

When I sent the print via OctoPrint it had extremely long pauses to the point I have to simply stop the print altogether. I then did a hard reboot (power on/off) to the printer and a full power cycle to OctoPrint. Sent it via OctoPrint and experienced the exact same issue.

I then exported directly into the MK3 SD card and the print worked without any issues or delays.

leshkinski, 2018-10-25:
Performance of the device OctoPrint runs on also affects this, so might be worth checking how much load it's dealing with when printing.

dikodahan, 2018-10-25:
My OctoPrint is running on a dedicated Raspberry Pi 3B. Nothing else is running on it.

stahlfabrik, 2018-10-25:
Isn't the Prusa Print Image optimized for the Pi zero? You could try a regular OctoPi image...

How is your Pi connected to the printer?

dikodahan, 2018-11-10:
PrusaPrint (that is actually an OctoPi image pre-configured for the Prusa Printers) can be installed on both Pi Zero and 3B, 3B+ models. That should have any impact on this issue.

With that, after much work on this, it seems like my RPi 3B board kicked the bucket... I replaced it with a new RPi 3B+ and it is now working again without any issues.
So, I would chock this one up to a hardware issue.

Thank you and you guys can close the case.

closed 2018-12-14 by michalxfanta