prusa3d/Prusa-Firmware #1270: Speed changes, sometimes random, sometimes matching "Retraction Speed" setting

AbeFM, 2018-10-19:
Frankly I'm tired of posting it in 20 places.

If someone here cares, let me know. Long detailed write ups with testing take a long time to write up, and they are generally ignored. Suffice it to say the print speed adjusts itself up and down by a factor of 3.

jonbet83, 2018-10-19:
Hi AbeFM, I’ve had my MMU for a while now and it’s been a bit hit and miss but generally works ok. I’ve been following MMU issues etc for a while and noticed you posting a lot, why don’t you come over to discord chat and we can try and help you?

jonbet83, 2018-10-19:
https://discord.gg/CQcc3Q

AbeFM, 2018-10-19:
I can't seem to post there.

It's certainly not about making me feel better. It's about making me think I'm not wasting my time drawing up test objects, trying settings back to back, developing new parts, editing firmware and g-code by hand for hours, finding multiple people who have the same issue and then.....

Crickets.

I would like to make the team aware that the printers are not using 100% FR, and that it varies - specifically, the speed will often set itself to the specific n-1 extuder's retraction speed.

I do not have permissions to post in the thread linked. I've never used discord: How do I write in the place you want me to?

jonbet83, 2018-10-19:
Discord is a chat server, make an account and that link should work. In all honesty I don’t see loads of people Having the issues like you have but I’m only speaking on my experiences and they are generally mostly good from Facebook and discord. G codes generated from slic3r do work fine and I’m able to produce prints to good quality also

AbeFM, 2018-10-19:
I produce plenty of good prints.

I hope it's clear that my github posts aren't intended to be "fix my print". It's why I don't post pictures of birds' nests and such, and instead try to post data, like "I set each extruder to a different retraction speed and found the print speed to vary 1:1 with those numbers".

And it's why I end such posts with "I'd be happy to try anything you need".

I would really like to see this issue of print speeds changing looked at, or at least acknowledged.

Does it sound like the sort of thing that would be a problem if it were true?

Would you want to prevent the printers from constantly changing their print speed?

Do you think I'm making it up? I would be happy to provide videos, code, etc. I will post it on discord but how will it help?

https://github.com/prusa3d/Slic3r/issues/1313 If you know where the RIGHT place is to post it, could you post it there and link to it so I can see my efforts invested where they will do some good instead of taking hours of your time telling me this or that isn't the right place? I don't want attention, I want to post the issue, the data, and then later the fix comes out - and I am willing to answer questions or experiment to assist.

jonbet83, 2018-10-19:
#wastingmybreath #nowondertheyignoreyou #goodluckandgoodbye

AbeFM, 2018-10-19:
Comically, you posted on a related topic which was also ignored - speeds changing with the MMU.

I like your approach, 20 days and more identical issues posted, cluttering up GIT with duplicate issues.

AbeFM, 2018-10-20:
Moderators and Developers:
What can I do to move this process *forward*? I would like to know if you are considering this, and what I could do to make that process easier for all involved.

AbeFM, 2018-10-23:
Version
1.41.1b (MMU)

Operating system type + version
win10x64

Behavior
Printing speed sets itself from 100% to Extruder Retraction Speed. Each time a filament change finishes, the printing speed sets itself to 35.
I changed the retraction speeds to (Ext1,2,3...) 36,37,38,39,33 respectively.
I came in and saw printer at 35%, overrode it to 100.
F2: 36%
F3: 37%
F4: ..
and the speed would be 35/36/37/38 as it progressed through the channels.

It seems it's grabbing the Retraction Speed from each Ext(n-1) and setting that as the speed when it is loading the filament.

I think that's a bug.

Is this a new feature request?
No

STL/Config (.ZIP) where problem occurs
Upload a zipped copy of an STL and your config (File -> Export Config)

SpeedsChanging.zip

closed 2023-09-06 by github-actions[bot]