prusa3d/Prusa-Firmware #1277: M913 Not printing running currents

knoopx, 2018-10-24:
`tmc2130_print_currents` method is commented out

PavelSindler, 2018-11-02:
These gcodes were used for our debugging/development purposes. We will change these gcodes little bit to follow standard. My collegue @XPila will look on it. Thank you for letting us know.

PavelSindler, 2018-11-12:
Standard M907 gcode can be now used for setting motor currents on MK3.
Non-standard obsolete gcodes M910 - M918 have been disabled. However these gcodes can be still used by  uncommenting line  #ifdef TMC2130_SERVICE_CODES_M910_M918 in Configuration_prusa.h and rebuilding. M913 lines were uncommented.
Check these pull requests:

https://github.com/prusa3d/Prusa-Firmware/pull/1305
https://github.com/prusa3d/Prusa-Firmware/pull/1320

closed 2018-11-12 by PavelSindler

vertigo235, 2019-05-15:
Are the M907 settings stored in EEPROM?  Is there a way to recover the currently set current?  If I restart my printer do the firmware default currents restore?

Shootquinn, 2019-06-30:
> Standard M907 gcode can be now used for setting motor currents on MK3.
> Non-standard obsolete gcodes M910 - M918 have been disabled. However these gcodes can be still used by uncommenting line #ifdef [...]

With M912 disabled, the community needs to know if M907 can be used in the same way as it is for the Mk2, or if this is still a scale factor from 0 - 62 as current is set in Mk3 firmeware Also, since the Mk2 can output higher currents from the A4982 than the Mk3's TMC2130 is capable of, a reasonable range of values for M907 should be specified.