prusa3d/Prusa-Firmware #1285: 3.4.1 Hotend crashes into bed when going through mesh calibration at the start of a print

Crunch69, 2018-10-28:
3.4.1 Hotend crashes into bed when going through mesh calibration at the start of a print in octoprint.  The only way to sometimes stop it is to do a auto home from the calibration menu prior to starting the print

inquisitiveKangaroo, 2018-10-31:
Hi,
have you tried to move your pinda probe slightly lower? 
Is it possible to share the gcode and a video of this behaviour? (every single detail might be important) .)

Crunch69, 2018-11-08:
Dont have to move the PINDA only have to change to older firmware and problem goes away

Jacky2k, 2018-11-19:
Same problem here. Cancleded a job using octoprint while he was calibrating. Started the job again after a couple of seconds and he crashed into the bed. After that he touched the bed while doing the wizard. Moving the pinda probe one quarter turn down solved the problem for me.

3d-gussner, 2020-07-13:
@Crunch69 Is this still an issue? If it has been fixed in a newer firmware version, please close this issue.
[stale]

JohnnyDeer, 2020-08-12:
I'm closing this issue as stale, thanks for understanding.

closed 2020-08-12 by JohnnyDeer