prusa3d/Prusa-Firmware #1289: MK3 on the X and Y axis I have visible vertical lines yet the surface feels smooth.

Crunch69, 2018-10-30:
MK3 on the X and Y axis I have visible vertical lines yet the surface feels smooth.   These lines show up on every print made on the MK3.  I have made prints on the MK3 and a CR-10 using the same gcode and the vertical lines only show up on the MK3?

inquisitiveKangaroo, 2018-10-31:
Hi,
thank you for your feedback.)
Would be possible to share the gcode and maybe even a photo. I would like to try if I can reproduce it by my own.

cjo20, 2018-11-01:
I wouldn't be surprised if it was this issue:
https://github.com/prusa3d/Prusa-Firmware/issues/428

Crunch69, 2018-11-08:
[image]
[image]
[image]

stahlfabrik, 2018-11-08:
Welcome to the club. For me this is 603 - meaning the next big issue (after 602) that the MK3 has. 

As far as I understand, this is the issue of the TMC2130 stepper drivers being not optimally tuned for YOUR (and mine) motors. 

Maybe the stepper linearity settings for X and Y would help here - but how to find the right settings? There is also one user on GitHub, that claims to have made the lines disappear. See #771

Crunch69, 2018-11-13:
Christoph I have tried to contact NordicGuy to see how he was able to determine the correct settings but had no luck getting a responce

stahlfabrik, 2018-11-13:
Hi Crunch! Yeah. I really hope he will see my cry for help:-)

What he explains sounds straightforward and manageable. But how is it done exactly?

HolgerHU, 2018-11-16:
dgsdg

HolgerHU, 2018-11-16:
I also have these vertical lines. Everyone will have them!
This line pattern changes depending on the selected resolution (0.2, 0.15, 0.1 ...)
In my opinion, they are a result of every extruder step. The extruder stepper pushes the filament into the hotend using the Bondtech gear, which in turn briefly emits a larger amount of filament, which ultimately leads to this wave pattern. The extruder should therefore be operated at the finest possible step resolution to make the finished output as uniform as possible. In every movement, X- as in Y-direction.

Crunch69, 2018-11-16:
HolgerHU extruder artifacts are horizontal not vertical see the 602 inconstant extrusion

Vertical lines or banding has something to do in firmware and the tmc2130 driver settings to match our specific stepper motors.  There is a inconsistency in the manufacture of the motors

we need a FW driver tuning fix for this issue say a range of settings we can plugin compile and try for each independent printer

roberodin, 2019-11-29:
Fuck!

closed 2023-09-14 by github-actions[bot]