prusa3d/Prusa-Firmware #1307: Filament Sensor on MK3 not working properly with black PETG

kkrcma, 2018-11-06:
Have installed firmware 3.4.1 on MK3 and the filament sensor works for various colors of PLA, however, when using black PETG from Matterhackers the print will occasionally stop and request filament change. No filament change is in the gcode.  The only way I can get PETG prints to complete is to disable the filament sensor.

hoppke, 2018-11-10:
For what it’s worth, I’ve had 0 problems with the filament sensor - on 3.4.1 I went through 1kg of eSun’s black PETG - no false alarms from the filament probe - then ran out of this black PETG mid-print (correctly detected by the probe), loaded some _transparent_ PETG (Surreal brand this time) and continued from there. I’m currently about 80% through the transparent PETG and so far no runout/poor quality warnings from the sensor either. So on my MK3 this one feature is flawless. Making me think that perhaps there is something in 3.4.1 that makes the sensor misbehave more often, but perhaps it only happens if your actual sensor is in some way sub-par, or there’s some misalignment/flaw in the extruder carriage that affects the light path? Maybe you could get Prusa to replace the sensor on warranty?

inquisitiveKangaroo, 2018-11-20:
"filament sensor response is poor, disable it?" #1178 

We are aware of unstable fil. sensor response to some kinds of materials. Surprisingly black PETG is one of the worst materials in the matter of reliability of reaction, even though it's neither transparent nor abrasiv. We tried to add a 'warrning feature' but it wasn't especially reliable neither so it was eventually removed. Those sensors are anything but one like another, also the room enlightenment makes significant difference which makes the tuning more difficult. We are still working on enhancement, however, it seems like some solution less reliable on the light pollution would be more practical. We appreciate your inputs. Such feedbacks are essential to us.

ps: Sry for not refering it

Pathogenius, 2018-11-21:
while you wait:

https://www.thingiverse.com/thing:3223513

dikodahan, 2018-11-21:
> while you wait:
> 
> https://www.thingiverse.com/thing:3223513

Holy moly!!!
Prusa team, can you guys check this? Looks like he found a mechanical way to resolve this issue, much like the PINDA sensor in MMU2 did.

marcopiraccini, 2018-12-02:
Hi, has this been fixed in 3.5.0? This issue is still open, but I will not upgrade the firmware until this is fixed...

rennne, 2019-01-31:
I have the same issue with
Filamentum - HPE HG100 Extrafill Black Soul

I bought it at the Prusa Shop

DRracer, 2020-09-22:
Is this issue present in the latest FW 3.9.1? FW 3.9.0 brought severe improvements of the original MK3 PAT9125 filament sensor.

closed 2023-09-06 by github-actions[bot]